haplokuon/netDxf #110: Viewport with ApplicationRegistries

zzippo-de, 2019-05-25:
If a Viewport uses ApplicationRegistries, they are not added to the ApplicationRegistries.References
Is that intended?

haplokuon, 2019-05-28:
The code is working as expected. You are adding extended data information to the layout.Viewport, this is the viewport with Id 1 that represents the paper space itself, it has no graphical representation, and does not show the model. Create a new viewport, add the extended data to it, and add it to the layout as with any other entity.

zzippo-de, 2019-05-30:
OK. thank you for the answer, I tried to use netDFX to clean up very old files, which have grown over the past decades. I thought I can delete the ApplicationRegistry entry, if the reference list is empty.
But than after saving, acad crashes because of the extended data in the layout viewport.

haplokuon, 2019-06-02:
Now I see where the problem is, and, this goes to everybody, don't be so sparse in words, it is important to explain what you were trying to do.

haplokuon, 2019-06-23:
Fixed in netDxf 2.3.0

closed 2019-06-23 by haplokuon