haplokuon/netDxf #113: DimLinePosition retrun a wrong result

kchbpy, 2019-06-27:
When read the LinearDimension which is rotated in 90° or 270° will get a wrong result.
Here is a pic
[image]

kchbpy, 2019-06-27:
[image]

haplokuon, 2019-06-27:
Where is the LinearDimension coming from? Are you using the latest netDxf release?

If it is coming from a DXF upload a sample file, I do not know which program are you using and I might not have it, I guess it is AutoCAD by the looks of the image.

If you are creating the LinearDimension programmatically post the code you are using to generate it.

For the time being, when your dimension line is parallel to the measured line, it is easier to use an AlignedDimension instead.

kchbpy, 2019-06-28:
The LinerDimension is made by AutoCAD2016
In the pic below left:CMD right:AutoCAD2016
[image]
And here is the dxf file

[ddd.zip](https://github.com/haplokuon/netDxf/files/3337365/ddd.zip)

haplokuon, 2019-07-01:
Ok, I uploaded a fix to the repository. It only affected Linear Dimensions loaded from a DXF and only under certain circumstances, it was related to the order of the reference points and the rotation. It should work know.

kchbpy, 2019-07-04:
Thumbs up to you

closed 2020-09-03 by haplokuon