haplokuon/netDxf #116: block load dxffile error

yonghuairuogu, 2019-07-29:
i want insert a entire dxf file to  another dxf file, when i load the dxffile as block,it appear the message ":“Closed polylines cannot be combined with other entities to make a hatch boundary path.”"

Block block = Block.Load(symblodxf);
 Insert insert = new Insert(block , new Vector2(Y, X));
dxf.AddEntity(insert);

[4302.zip](https://github.com/haplokuon/netDxf/files/3441055/4302.zip)

haplokuon, 2019-08-08:
The origin of this problem is that associated hatches are not supported when its boundary path is made of intersected entities. This requires to calculate the intersection points between the different entities, something that is not implemented.

At the moment you have two choices, make your hatches non-associative or make sure that the associated entities correspond to the actual boundary path of the hatch.

I will try to make a fix so at least that error does not appear, like when it is loaded as a document. It will still be useful even though if the internal data of the hatch needs to be updated the resulting file might give problems when loaded elsewhere.

yonghuairuogu, 2019-08-12:
I would appreciate it if you could fix  this program.because this file is a symbol of pipeline ,and it is formulated by institute, i cannot change its  composition.

haplokuon, 2019-08-16:
Try it now with the latest code, you will need to compile it. The block will be created, even thought any Hatch entity will have its associative property set to false.

closed 2020-09-03 by haplokuon