haplokuon/netDxf #118: tolerance value

Marcross91, 2019-08-01:
Goodmorning all,

i have some problem with getting tolerance value from a dimension.
I wrote this simple code in a button click with c#:

`openFileDialog1.ShowDialog();
            DxfDocument fileDXF = DxfDocument.Load(openFileDialog1.FileName);

            foreach (netDxf.Entities.Dimension quote in fileDXF.Dimensions) {

                listBox1.Items.Add(quote.DimensionType);
                listBox1.Items.Add(Math.Round(quote.Measurement, 2));
                listBox1.Items.Add(quote.Style.Tolerances.UpperLimit.ToString());
            }`

the listbox1 text is right for DimensionType and for Measurement but tolerance are ever equals to zero…

what am I doing wrong?

haplokuon, 2019-08-06:
By default the dimension style tolerance upper and lower limit are zero. But if you have manually changed the dimension measurement tolerance values I will need to take a look at your DXF. Reading and writing those values works as expected.

Marcross91, 2019-08-06:
Yes i manually changed the tolerance value.
This is the trial sample dxf file that i used for my little project with netDxf library.

[Trial.zip](https://github.com/haplokuon/netDxf/files/3473099/Trial.zip)

Thank you

haplokuon, 2019-08-08:
You are trying to read the tolerance limit from the dimension style, but you have, actually, modified it individually. Therefore the information you seek is in the StyleOverrides list of the dimension.
```c#
double upperLimit = (double) dxfDocument.Dimensions.ElementAt(3).StyleOverrides[DimensionStyleOverrideType.TolerancesUpperLimit].Value;
```
In any case, I found a bug in how the tolerance display method of a dimension style override is saved to a file, I will fix that.

Marcross91, 2019-08-08:
It works perfectly,

Thank you a lot

closed 2019-08-08 by Marcross91