haplokuon/netDxf #119: ¿Como agregar una nueva escala al documento?

jrosalesmeza, 2019-08-08:
Hola, quisiera saber si es posible agregar una nueva escala a un documento dxf y agregar dicha escala a cada elemento creado.
Y tengo otra duda, es posible cambiarle a un Layer la propiedad Linetype scale?

Espero una respuesta, muchas gracias.

kchbpy, 2019-08-08:
I'm a passer-by. Here's my personal opinion.
Q1：
No , you should deal with the scale things. 
You can traversal all `EntityObject` then use the `transformby` method to scale all of your entities .
Q2:  
Yes, you can build your linetype by using `LinetypeSegment` so you can do anyting to the line.

jrosalesmeza, 2019-08-08:
Hola, creo que no me hice entender, a la primera pregunta me refiero a esto.
[image]

jrosalesmeza, 2019-08-08:
Hola, creo que no me hice entender, me refería es a crear una escala tipo como se muestra en la imagen.
[image]

kchbpy, 2019-08-08:
Sorry, I don't quite know that.

haplokuon, 2019-08-10:
Puedes asignar manualmente la escala de impresión en "PlotSettings" de cada "Layout" pero definir las escalas de una lista, no.

En cuanto a las escalas de los tipos de línea, se controlan de dos formas mediante las variables globales "LtScale" y "PsLtScale" del dibujo o individualmente para cada entidad en "LinetypeScale".