haplokuon/netDxf #120: The StyleOverrides of Dimension not work

kchbpy, 2019-08-08:
Hi there:
    I use StyleOverrides to add tolerance but it is not work, I can't see the tolerance in AutoCAD. I also try to override the textHeight, still not work. I printout the `StyleOverrides.count` get right number.
    So I think it not write correctly to dxf file.

haplokuon, 2019-08-10:
The StyleOverrides works but it doesn't mean that a problem with one or more of them might appear. Perhaps, you are having problems with a bug I found with the tolerance display method variable when a file is saved while checking this [issue](https://github.com/haplokuon/netDxf/issues/118).

It would be a lot more productive if you actually show a piece of code so I can reproduce your problem.

kchbpy, 2019-08-10:
Here is the code:

```
using System;
using System.Collections.Generic;
using System.Linq;
using System.Text;
using System.Threading.Tasks;
using netDxf.Entities;
using netDxf;

namespace styleoverrideBug
{
    class Program
    {
        static void Main(string[] args)
        {
            DxfDocument doc = new netDxf.DxfDocument(netDxf.Header.DxfVersion.AutoCad2010);
            netDxf.Collections.DimensionStyleOverrideDictionary overrides;
            AlignedDimension ad = new AlignedDimension(new Vector2(0, -30), new Vector2(50, -35), 10);
            overrides = ad.StyleOverrides;
            overrides.Add(
                netDxf.Tables.DimensionStyleOverrideType.TolerancesDisplayMethod,
                netDxf.Tables.DimensionStyleTolerancesDisplayMethod.Symmetrical
                );
            overrides.Add(
                netDxf.Tables.DimensionStyleOverrideType.TolerancesLowerLimit,
                0.2
                );
            overrides.Add(
                netDxf.Tables.DimensionStyleOverrideType.TolerancesUpperLimit,
                0.1
                );
            overrides.Add(netDxf.Tables.DimensionStyleOverrideType.TolerancesSuppressLinearTrailingZeros, true);
            doc.AddEntity(ad);

            doc.Save("test.dxf");
        }
    }
}
```

And this is the result:

[image]

haplokuon, 2019-08-12:
As I imagined you were suffering the issue I pointed out in my previous comment. It should be fixed now with the latest commit.

On the other hand, you are defining the display method as Symmetrical and, later, you are assigning different values for the upper and lower limit. In such cases, the limits will take precedence and when the file is loaded again, it will show as Deviation.

kchbpy, 2019-08-14:
Thanks a lot.

closed 2020-09-03 by haplokuon