haplokuon/netDxf #121: LinkAssemblies error with Xamarin

jribault, 2019-08-09:
Hi,

I used with success netDxf on Debug with Xamarin Android, but now I try to build the project in Release mode I got the following LinkAssemblies error. The only difference between Debug and Release is the use of "Shared Runtime" which is not permitted in order to put an app on the playstore. Do you have an idea of what's wrong ?

Best regards,

Severity	Code	Description	Project	File	Line	Suppression State
Error		The "LinkAssemblies" task failed unexpectedly.
Mono.Linker.MarkException: Error processing method: 'System.String netDxf.Tables.TextStyle::TrueTypeFontFamilyName(System.String)' in assembly: 'netDxf.netstandard.dll' ---> Mono.Cecil.ResolutionException: Failed to resolve System.Drawing.Text.PrivateFontCollection
   at Mono.Linker.Steps.MarkStep.HandleUnresolvedType(TypeReference reference)
   at Mono.Linker.Steps.MarkStep.MarkType(TypeReference reference)
   at MonoDroid.Tuner.MonoDroidMarkStep.MarkType(TypeReference reference)
   at Mono.Linker.Steps.MarkStep.MarkMethodBody(MethodBody body)
   at Mono.Linker.Steps.MarkStep.ProcessMethod(MethodDefinition method)
   at Mono.Linker.Steps.MarkStep.ProcessQueue()
   --- End of inner exception stack trace ---
   at Mono.Linker.Steps.MarkStep.ProcessQueue()
   at Mono.Linker.Steps.MarkStep.ProcessPrimaryQueue()
   at Mono.Linker.Steps.MarkStep.Process()
   at MonoDroid.Tuner.MonoDroidMarkStep.Process(LinkContext context)
   at Mono.Linker.Pipeline.Process(LinkContext context)
   at MonoDroid.Tuner.Linker.Process(LinkerOptions options, ILogger logger, LinkContext& context)
   at Xamarin.Android.Tasks.LinkAssemblies.Execute(DirectoryAssemblyResolver res)
   at Xamarin.Android.Tasks.LinkAssemblies.Execute()
   at Microsoft.Build.BackEnd.TaskExecutionHost.Microsoft.Build.BackEnd.ITaskExecutionHost.Execute()
   at Microsoft.Build.BackEnd.TaskBuilder.<ExecuteInstantiatedTask>d__26.MoveNext()	ArpentGIS.Android

haplokuon, 2019-08-10:
I do not use Xamarin but your problem seems similar as other issues people were having, it seems related with Mono and the PrivateFontCollection class of the net Framework. I have no idea why, but something you can try is to download the latest code and delete the method "TrueTypeFontFamilyName" of the TextStyle class. It can be done safely since that method is not used anymore, and the PrivateFontCollection class is only used there. I just left it just in case.

jribault, 2019-08-19:
Thanks it worked !