haplokuon/netDxf #122: Issues importing DXF Since 2.3.0 Update

Pietervdwest, 2019-08-22:
Hi we are experiencing some issues importing some DXF files into a Trutops cnc software. This issue has begun since the 2.3.0 update. It does not happen to all files only some of the files created.
My process:
Export a DXF file from Autodesk Inventor
Open DXF file - NetDXF
Edit some layers and colours - NetDXF
Add some text - NetDXF
Save DXF File - NetDXF

The issue I have is trying to import the file in Trumpf's Trutops CAD software.

haplokuon, 2019-08-28:
I tried your DXF files and I haven't found any issues with them. You are talking about issues when trying to load a file with Trumpf's Trutops CAD software, but you forgot to describe what are those issues. Perhaps it is specific for that software, I do not have it and I cannot test it. Try it with a different program to see if it shows the same problems, if it doesn't most probably there is something that your software doesn't like. I would recommend to test it with Autodesk DWGTrueView, it is free and since DXF file format is from them it should be consider as the standard to follow.

Try to pinpoint the modification you are doing that shows the issue and post the exact code and file you are using, better if you start with an empty drawing.

Pietervdwest, 2019-08-29:
Hi Daniel,

Thanks for getting back to me. Yes look the only thing I can think of is that it’s a shortcoming in their software because AutoCAD and DWGTrueview and a few other CAD packages I tried opens the new files just fine. Only reason why I asked is because the issue appeared after I migrated to the new netDxf.netstandard.dll version 2.3.0. So far its only the one group of people using Trumpf’s Trutops CAD application that’s got an issue.
When I was using the older netDxf.dll version 2.2.0 all was fine and even Trutops CAD had no problem opening the files generated.

As I said I can open all the files on my side with the CAD applications I use so I am not sure why there is now an issue. The error then is also not very specific. Only says unspecified error.

Thanks for your response and great work!

Kind regards,
Pieter van der Westhuizen
Technical Director

[Micrographics Final logo_Vertical_Rev1]
Tel: +27 41 365 5701
Mobile: +27 83 270 3525
Fax: 086 658 2289
Email: pieter@mgfx.co.za<mailto:pieter@mgfx.co.za>
Web: www.mgfx.co.za<http://www.mgfx.co.za/>

[MGFX EMAIL 2018 (002)]

This e-mail message is intended solely for the use of the addressee and may contain legally privileged and confidential information. If you are not the intended recipient or his/her representative, please be advised that any dissemination, distribution, copying or the use of this message or its attachments is strictly prohibited under law. If you have received this message in error, please notify the sender immediately and please delete this message and all attachments from your computer. All quotes are provided in good faith and subject to change without notice, and further subject to Micrographics terms and conditions, which are available upon request and can be found at www.mgfx.co.za/support/micrographics-terms-and-conditions/. Any information passed along in good faith from sub-consultants or advisors beyond the control of Micrographics is strictly to provide the addressee with an indication of the sub-consultants expertise and/or advice and Micrographics may not be held responsible for any commitments made that fall outside of its core function as an Autodesk Reseller. Should you have any doubt as to the content of this email, please immediately refer it to Micrographics at info@mgfx.co.za.

haplokuon, 2019-08-30:
If you are using a "netDxf.netstandard.dll", it is coming from another user that modified the project to use the .NET standard library instead of the .NET Framework 4.5 that I use. I really do not believe that such modification is the reason of your problem, but when reporting a possible bug, I recommnd to always use the latest dll published here at GitHub.

You will need to find manually what is generating that abnormal behavior of your software, I cannot do much about it. Start with an empty file generated by netDxf, do not load any other file, do not add anything, just save it, then you can try to start adding stuff to it until it breaks. The only idea I can give you is to start with adding TextStyles, try both main constructors the one that takes a font file name "public TextStyle(string font)" and the one that takes a font family and a font style "public TextStyle(string fontFamily, FontStyle fontStyle)". I have been working in them not long time ago, and if your software does not read the extended data information when reading the text style that takes the font family and the font style, it might fail.

Pietervdwest, 2019-09-05:
Thanks for you feedback and help. I will use netDXF.dll again.

Pietervdwest, 2019-09-05:
Hi Daniel I cant seem find the netdxf.dll version 2.3 to download. Nuget only still has version 2.2. Can you help?

This e-mail message is intended solely for the use of the addressee and may contain legally privileged and confidential information. If you are not the intended recipient or his/her representative, please be advised that any dissemination, distribution, copying or the use of this message or its attachments is strictly prohibited under law. If you have received this message in error, please notify the sender immediately and please delete this message and all attachments from your computer. All quotes are provided in good faith and subject to change without notice, and further subject to Micrographics terms and conditions, which are available upon request and can be found at www.mgfx.co.za/support/micrographics-terms-and-conditions/. Any information passed along in good faith from sub-consultants or advisors beyond the control of Micrographics is strictly to provide the addressee with an indication of the sub-consultants expertise and/or advice and Micrographics may not be held responsible for any commitments made that fall outside of its core function as an Autodesk Reseller. Should you have any doubt as to the content of this email, please immediately refer it to Micrographics at info@mgfx.co.za.

haplokuon, 2019-09-09:
You can find the latest netDxf.dll here at GitHub in the releases tab. I do not use and do not publish anything through Nuget.

Pietervdwest, 2019-09-10:
Thanks for the clarification that helps a lot to know because now I know I should get the dll from Github and not from Nuget.

Do I have to compile the source myself to get the latest updates as per you Changelog which reflects the latest change to be on (### [2019/08/30])?

Also is R12 DXF support completely removed?

This e-mail message is intended solely for the use of the addressee and may contain legally privileged and confidential information. If you are not the intended recipient or his/her representative, please be advised that any dissemination, distribution, copying or the use of this message or its attachments is strictly prohibited under law. If you have received this message in error, please notify the sender immediately and please delete this message and all attachments from your computer. All quotes are provided in good faith and subject to change without notice, and further subject to Micrographics terms and conditions, which are available upon request and can be found at www.mgfx.co.za/support/micrographics-terms-and-conditions/. Any information passed along in good faith from sub-consultants or advisors beyond the control of Micrographics is strictly to provide the addressee with an indication of the sub-consultants expertise and/or advice and Micrographics may not be held responsible for any commitments made that fall outside of its core function as an Autodesk Reseller. Should you have any doubt as to the content of this email, please immediately refer it to Micrographics at info@mgfx.co.za.

haplokuon, 2019-09-10:
There is no need to compile the latest code just use the compiled library that you can download in the releases tab. You can still use the libraries you can find at Nuget, but keep in mind that I do not maintain those.

No, any DXF version prior to AutoCAD2000 database is not supported.

Pietervdwest, 2019-09-11:
Hi Daniel,

Ok after a lot of digging for what could be causing this I have narrowed it down to one thing. The issue is caused by splines created by netDXF. Let me just quickly explain my workflow and what I am doing.

My application is basically a post-processing application which opens DXF files created by Inventor which is an application from Autodesk. Once the DXF file has been opened it adds text or a few lines and then its saved again with the same filename.

I have studied these files and working with the user who has the issue importing the files I could narrow it down to the splines inside the DXF file. I then had a closer look at the individual spline properties inside AutoCAD and I could see that after the file was processed (Opened and saved) using netDXF function, all the splines control point weights were changed to 1. In the original file the control points of all splines are set to -1. Even if I create a new spline inside the DXF file using AutoCAD the control points weight is -1.

I then did a test using AutoCAD where I opened the original DXF file and saved it again as the same version DXF and the spline control points stayed on -1.

So all I can tell you is that if a DXF file is processed by a netDXF function the resulting file splines control points are all set to 1 even though they were -1 to begin with.

Please see files and images here: https://www.dropbox.com/s/83g2nianjr5kcbb/DXF%20Issue.zip?dl=1

Thanks for the help thus far and I hope to hear from you soon.

Pieter
This e-mail message is intended solely for the use of the addressee and may contain legally privileged and confidential information. If you are not the intended recipient or his/her representative, please be advised that any dissemination, distribution, copying or the use of this message or its attachments is strictly prohibited under law. If you have received this message in error, please notify the sender immediately and please delete this message and all attachments from your computer. All quotes are provided in good faith and subject to change without notice, and further subject to Micrographics terms and conditions, which are available upon request and can be found at www.mgfx.co.za/support/micrographics-terms-and-conditions/. Any information passed along in good faith from sub-consultants or advisors beyond the control of Micrographics is strictly to provide the addressee with an indication of the sub-consultants expertise and/or advice and Micrographics may not be held responsible for any commitments made that fall outside of its core function as an Autodesk Reseller. Should you have any doubt as to the content of this email, please immediately refer it to Micrographics at info@mgfx.co.za.

haplokuon, 2019-09-19:
The control point weight value cannot be your problem it has been like that since, I guess, the day I implemented the Spline entity.

For reasons I don't know, AutoCad initially sets the control points weight as -1 when a new Spline is created. But it is the same as it was using 1. You can easily test this. Create a Spline using the CV method, and the try to modify the weight of one of its control points, set it to 1 and you will not see any change in the curve. Later if you try to set the same weight to a negative number it will not allow you.

I just know the basics about NURBS but it seems weird to have negative weights, or even zero, In all the documentation I read about NURBS I have never seen an example of negative weights. AutoCad does not allow them, if I remember correctly Rhinoceros does not either. Therefore to avoid problems when a DXF is loaded with netDxf any negative weights will be set to 1.

Pietervdwest, 2019-09-19:
Hi Daniel,

Thanks for getting back to me.

I understand what you mean. I did isolate the issue to be relating to the spline entities created so that was why I was looking if I could see any differences between the two files when opening them in AutoCAD. It was only the weight on the control points I could see was different hence why I mentioned it to you.

If I change the geometry in the originating CAD program so that there are no complex geometry (splines) and export the DXF it works 100% after it has been taken though .netDXF processing and the final CAD product can import the file without any issues. So all I can tell you is that my issue is related to splines being processed by .netDXF.

Like I said, even if I just open the DXF file (which contains spline entities) and save it again using .netDXF then it can’t be imported any longer. If there are no splines involved it works fine.

What do you suggest I try next as I would like to get this solved…

This e-mail message is intended solely for the use of the addressee and may contain legally privileged and confidential information. If you are not the intended recipient or his/her representative, please be advised that any dissemination, distribution, copying or the use of this message or its attachments is strictly prohibited under law. If you have received this message in error, please notify the sender immediately and please delete this message and all attachments from your computer. All quotes are provided in good faith and subject to change without notice, and further subject to Micrographics terms and conditions, which are available upon request and can be found at www.mgfx.co.za/support/micrographics-terms-and-conditions/. Any information passed along in good faith from sub-consultants or advisors beyond the control of Micrographics is strictly to provide the addressee with an indication of the sub-consultants expertise and/or advice and Micrographics may not be held responsible for any commitments made that fall outside of its core function as an Autodesk Reseller. Should you have any doubt as to the content of this email, please immediately refer it to Micrographics at info@mgfx.co.za.

haplokuon, 2019-09-25:
I do not have much to tell you. I tested your DXF files with different software and none showed any problem. Have your tried what I was recommend you on my second post? Try it with a new file, do not load an external one, add an spline to the document, and try to load the result with your software.

Regardless of what your problem might be, the way the Spline is saved in the DXF has not changed with 2.3.0. You can try the latest commit, download it compile it, and test it with that. You can also go through the changes introduced in 2.3.0 to see if one of the changes might affect you.

Pietervdwest, 2019-09-25:
Hi Daniel.
I am inContact with the Trumpf supplier (https://www.trumpf.com) here in my country and will hopefully be getting a trial copy of their CAD software to see if I can isolate the issue some more. I will give you more feedback soon. Thanks again for the help.
This e-mail message is intended solely for the use of the addressee and may contain legally privileged and confidential information. If you are not the intended recipient or his/her representative, please be advised that any dissemination, distribution, copying or the use of this message or its attachments is strictly prohibited under law. If you have received this message in error, please notify the sender immediately and please delete this message and all attachments from your computer. All quotes are provided in good faith and subject to change without notice, and further subject to Micrographics terms and conditions, which are available upon request and can be found at www.mgfx.co.za/support/micrographics-terms-and-conditions/. Any information passed along in good faith from sub-consultants or advisors beyond the control of Micrographics is strictly to provide the addressee with an indication of the sub-consultants expertise and/or advice and Micrographics may not be held responsible for any commitments made that fall outside of its core function as an Autodesk Reseller. Should you have any doubt as to the content of this email, please immediately refer it to Micrographics at info@mgfx.co.za.

donnyv, 2019-12-31:
2 things I learned from this issue.
@haplokuon doesn't support the nuget dlls and did not do the "netDxf.netstandard.dll" version

Is there a reason you don't support a nuget install? It pretty much the standard way .net developers install libraries now.

Pietervdwest, 2020-01-02:
@donnyv  I agree that would be a nice option to be able to get a signed package of @haplokuon  latest build as a Nuget package.