haplokuon/netDxf #124: DimensionBlock.Build() get wrong result

kchbpy, 2019-08-25:
I readed a dxf file named "ttt.dxf" then build the `DimentionBlock` and move the block down saved as "test.dxf".
Here is the code:
`class Program
    {
        static void Main(string[] args)
        {

            DxfDocument doc = DxfDocument.Load("ttt.dxf");

            List<EntityObject> ens = new List<EntityObject>();

            foreach (var item in doc.Dimensions)
            {
                //***********Use the "DimensionBlock.Build" while get wrong result**********
                netDxf.Blocks.Block bb = DimensionBlock.Build(item);
                //*******If use the block in file I can get right result********
                //netDxf.Blocks.Block bb = item.Block;
                foreach (var tt in bb.Entities)
                {
                    EntityObject en = (EntityObject)tt.Clone();
                    en.TransformBy(Matrix4.Translation(0, -50, 0));
                    ens.Add(en);
                }
            }
            doc.AddEntity(ens);
            doc.Save("test.dxf");
            Console.ReadKey();
        }
    }`

This is the "ttt.dxf":
[image]

This is the "test.dxf":
[image]

The files

[files.zip](https://github.com/haplokuon/netDxf/files/3537911/files.zip)

haplokuon, 2019-08-28:
All dimensions contains a reference to a block that holds its drawing representation. If this value is set null it will force the program that reads the resulting DXF file to generate the dimension drawing block, some programs do not even care about this block and will always generate their own dimension drawings. Therefore there is no need to build those blocks.

In the past I needed some basic dimension drawing, that is why that code exists, I left it there in case someone, exceptionally, needs it. The actual code does not use the full set of parameters defined in the dimension styles and I have no plans to update it. Except to fix problems like those two dimensions on the bottom right of the image that have the reference lines crossed.

So, if you are loading a DXF file with dimensions that you plan to modify, I would recommend to set the dimension.Block property to null. From your code if you do not update the block everything looks correct, the block from the original dimension looks fine since it has not been modified; and when you clone a dimension its associated block is not cloned with it, it will be set to null, and when loaded again with AutoCad it will be rebuild.

kchbpy, 2019-08-30:
I want to trans the dxf to image, I finished the `Line` `MTex`t `Circle` `Solid`...... 
So,I want use the `Entities` in `Dimensions` to convert to image.
Or,is there a method to explode the `Dimensions`?

haplokuon, 2019-08-30:
I have been unable to reproduce the problem with the two dimensions on the bottom right of your image with the crossed reference lines, make sure you are using the latest version. I remember fixing something like this sometime ago.

I do not know exactly what are you trying to achieve with "netDxf.Blocks.Block bb = DimensionBlock.Build(item);". If for any reason you need to uses the built in dimension drawing the correct way of doing it is directly through the Block property of the dimension like
```c#
foreach (Dimension dim in doc.Dimensions)
{
	Dimension en = (Dimension)dim.Clone();
	en.TransformBy(Matrix4.Translation(0, -50, 0));
	en.Block = DimensionBlock.Build(en);
	ens.Add(en);
}
```
or you can set to true the document property "BuildDimensionBlocks" so when a dimension is added to the document, the block that represents the dimension drawing will be automatically generated. But, keep in mind that the dimension drawing code is limited, even I have found that when working with AutoCad it will always regenerate the drawing of the Angular3PointDimension not respecting the block stored in the DXF.

If you need to duplicate a dimension without having to rebuild the block, remember, that block is just like any other block, you can clone it with another name and use it to create an insert and transform that, or you could make a copy of the entities contained inside a block and transform them, instead. The entities stored inside the block of a dimension are expressed in world coordinates.
```c#
foreach (Dimension dim in doc.Dimensions)
{
	foreach (EntityObject o in dim.Block.Entities)
	{
		EntityObject clone = (EntityObject) o.Clone();
		clone.TransformBy(Matrix4.Translation(0, -50, 0));
		ens.Add(clone);        
	}
}
```

closed 2020-09-03 by haplokuon