haplokuon/netDxf #127: Error reading DXF

johnnyontheweb, 2019-09-06:
Hi, 
I have a DXF file that opens correctly in Brics, nanocad, etc... but it fails opening in AutoCAD, and in netDxf too.
In the line of errors I read the code 5005. What is this?

`...
5005
bpenEnabled
290
     1
5005
bpenFileName
5005
sketchy.bpen
  0
XRECORD
  5
....`

haplokuon, 2019-09-09:
The DXF code 5005 does not exist, at least the largest code that appears in the official DXF documentation is 1071; and, besides other possible problems outside what you are showing here, is the reason why netDxf is not opening the file, and most probably for AutoCad too You cannot invent your own DXF codes and expect that everything will be alright.

Without what comes before the line you showed, I have no idea what the program was trying to do when wrote that, l guess it is some specific data of the program that wrote that. Who knows.

johnnyontheweb, 2019-09-09:
Thanks for your reply. IMHO, you cannot assume that the library is limited to autodesk DXF files. All the other CAD programs (Brics, nanoCAD, etc... ) opens the attached file. Autocad and your library don't.

In general, I have some troubles with line 310 in /io/TextCodeValueReader.cs, which throws an exceptions if something's unknown. I humbly suggest to remove the exception there - is it ok if ReadValue returns nothing?
[test.zip](https://github.com/haplokuon/netDxf/files/3591440/test.zip)

haplokuon, 2019-09-10:
The DXF is not an open file format, it was developed by Autodesk for AutoCAD many years ago, therefore there is only one type of DXF and, for good or bad, it is Autodesk DXF. I am not assuming anything it is just the way it is. It doesn't mater if every single program is able to read that file, if AutoCAD or one of its acolytes cannot, the problem is with the file.  If you ask my opinion the DXF format is nowadays a complete bloated mess.

And now some basic information about the DXF format. The most basic piece of data in a DXF is composed by a Code and a Value. The code defines the way the Value is read, and the Value holds the actual data. Also a DXF can be stored as a text or a binary file, and both are saved in exactly the same way.

In a text file the Code and Value information are stored in different lines, first the Code is read and once we have identified the kind of value that is stored in the next line, we can convert it to a short, an integer, a float, a double, a string, a hexadecimal number, whatever. If we cannot recognize the Code type there is now way we can read the Value data. We could just take the naive solution and skip the line since in a text DXF Codes and Values are stored in different lines, but the problem persists we cannot read the Value line and we will never know how important it was.

Now in a binary DXF this problem is much, much worse making the file completely unreadable. In a binary file there are no carriage return or line feed characters that separates Codes and Value pairs, only a bunch of consecutive bytes. If we arrive to an unrecognizable Code, we know that it is two bytes and the Value? who knows 1, 2 4, 8,.., get the number of bytes wrong, and the DXF binary reader falls apart. We will never know where the next Code starts.

Summarizing, the program that created that file has a serious problem. I do not care if there are programs that are able to read such a file, this is not a bug with netDxf nor with AutoCAD, as I said in my previous post "you cannot invent your own DXF codes and expect that everything will be alright".

Your only solution with that file is to manually replace that infamous 5005 codes for a code that actually exists and that holds a string since it is what appears after that; choose one of the available codes and hope for the best.

johnnyontheweb, 2019-09-10:
Thanks for the clear reply

closed 2019-09-10 by johnnyontheweb