haplokuon/netDxf #128: Block Description Doesn't Write To Save

alexeiholgate, 2019-09-17:
I've edited the description for a block. When saving and opening in AutoCAD, this description is lost. I also cant find it in Notepad++. I can successfully write to Name but not Description.

haplokuon, 2019-09-19:
Yes, I forgot to add it. And now I remember why. There is a but with AutoCad related to this, it will allow you to create blocks with very long descriptions, and if you try to load a DXF file with them it will show an error. I do not know the exact limit but it is around the 3500 characters plus minus no idea. 2048 is fine 4096 too many.

Anyways I will add it, it will be your responsibility to avoid problems like this or the use of new line characters.

alexeiholgate, 2019-09-19:
No worries, descriptions shouldn't be longer than 200 characters in my case.

closed 2020-09-03 by haplokuon