haplokuon/netDxf #129: Write bezier to dxf

fleixi, 2019-09-19:
I have some bezier curve created from cspotrace ([cspotrace](https://gitlab.com/orobert/cspotrace/blob/master/CsPotrace.cs)) and want to write them to a dxf file but i have no clue how to convert it.

I have tried to use polyline and set the smoothstyle but this is not changeable and set to nosmooth and i have tried spline but a spline is no bezier and the output do not match.

Are there a way to convert bezier to something matching in netDxf

OakLD, 2019-09-27:
While the author could provide a definite answer, I would say that there is nothing in netDxf that might be of a help. There are two good reasons for that: Beziers are not used much in CAD software (to my best knowledge), due to their nature, they are used in different kind of modeling (inexact, for presentation). Second reason is, that it is defficult to convert Bezier to Splie. Here is some info on the subject: [how-can-i-convert-beziercurve-to-b-spline](https://stackoverflow.com/questions/27476796/how-can-i-convert-beziercurve-to-b-spline)

wieslawsoltes, 2019-09-27:
That's what I use:
https://github.com/wieslawsoltes/Core2D/blob/84b575a238f688c4f7d531728abd362a578f4527/src/Core2D.Renderer.Dxf/DxfRenderer.cs#L177-L196
https://github.com/wieslawsoltes/Core2D/blob/84b575a238f688c4f7d531728abd362a578f4527/src/Core2D.Renderer.Dxf/DxfRenderer.cs#L357-L371