haplokuon/netDxf #130: How to create an image with relative path

lijiayi0921, 2019-09-23:
I am trying to add an image with relative path with netDxf.
However it seems impossible to create an ImageDefinition in this way, since in the ctor it says "Image file not found".
And I cannot modify the "File" field afterwards.
What's the best workaround for this ?
Thank you

haplokuon, 2019-09-25:
I cannot see any problems passing a relative path to the ImageDefinition constructor. It would have been a lot more useful if you would have shown what your actual code looks like.

lijiayi0921, 2019-09-26:
@haplokuon 

Thank your for your reply.
I am using netDXF 2.2.0.1 from NuGet.
Here is my code :
                **ImageDefinition imageDefinition = new ImageDefinition(imageName, imageFile);**
If I define imageFile as @".\XREF\xxx.jpg", where XREF foldler is located besides the target dxf doc.

haplokuon, 2019-09-27:
Now I understand your issue, there is no such thing as too much information, the evil is in the details.

There are two main types of constructors for the ImageDefinition. The generic one that besides needing the name and the file of the image, it requires the dimensions, the resolution, and the units. These constructors will not check if the image actually exists since all information required by the image definition is already passed in the constructor.

On the other hand the two other simplified constructors that only require the name and the file, needs to find the image in order to get the rest of the data. When you initialize an ImageDefinition it does not know anything about the future location of the DXF, but it still needs to find the image file. If a relative path is used it will look for it following the actual System.Environment.CurrentDirectory, therefore you will need to set this variable to where you are planning to save the DXF, before creating a new ImageDefinition.

Choose one of the two options.

lijiayi0921, 2019-09-27:
Hi haplokuon,
Thank you for your explanation.
The second option suits better in my case since I need the basic information of the given image to fit it into my display zone.
I will try it next Monday and let you know if everything works fine.

lijiayi0921, 2019-09-30:
Hi @haplokuon ,
I am experiencing some new problem.
With your suggestion No. 1 (relative path with System.Environment.CurrentDirectory), the ImageDefinition and Image are successfully created, and DXF well generated.

How ever when opened in AutoCAD (2017 and 2019), the image content is not shown. Only the frame is shown:
[image]

With image Misc information like this
[image]

In the "xref" dialog, images are shown as "unreadable".
If I make the path "absolute", image is loaded. If I then change it back to "relative", image is still fine. Meanwhile the "Saved Path" and other properties remain identical to when it was unreadable.
[image]

More strangly, even if the image is marked "unreadable", in the Brightness/Contrast/Fade dialog, the preview is fine.
[image].

The same dxf works perectly in MicroStation.

Do you have any idea about how to solve this ?
Thank you in advance.

Jiayi

haplokuon, 2019-10-01:
I am aware of this issue, and I cannot do anything about it. It seems that AutoCAD doesn't like relative paths when inserting an image in a drawing. However, it has an easy workaround, in AutoCAD go to the "External References Palette" and "Reload All References", the images should appear, it knows where to find the files but for some reason they do not appear.  It looks like one of those everlasting bugs in AutoCAD, that they never bothered to fix.

lijiayi0921, 2019-10-02:
Thank you for the workaround!
Cheers

philippePrevu3D, 2021-10-04:
Hi, Sorry to open this thread again. I faced the same issue with Relatives path.

I got an answer from AutoCAD staff that could help you face the problem. 
It seem like the images should be embedded in the DXF. At the moment, is it possible to do it with netdxf?

See my discussion below with Autocad:
[image]
the links he sent me : 
[](https://gis.stackexchange.com/questions/97455/can-a-dxf-file-contain-raster-images-tiff-jpg)

philippePrevu3D, 2021-10-06:
Thanks a lot for your reply @haplokuon 
I will honestly forward that answer and keep you aware of the result. (I might censor some part of it just to make sure to have an answer from the guy, took me around 2 weeks to finally get in touch with someone that might be able to solve that situation) or at least try