haplokuon/netDxf #138: DxfDocument is null when loading an 2007 DXF

Mohammed001, 2019-10-20:
I wonder why is this happening with my file (https://drive.google.com/file/d/1kjayAdpJetv10ZCs0GrlgheJtwu7y45Y/view?usp=sharing) which i've been created using LibrCAD program!

```
DxfDocument loaded = DxfDocument.Load(file);
if (loaded == null) Console.WriteLine("error");
```

any help, please

haplokuon, 2019-10-23:
Your are suffering this same [problem](https://github.com/haplokuon/netDxf/issues/105).

This is a bug with the LibreCad dxf writer.

mlkmlt, 2019-10-25:
Hey,
I also use LibreCad and solved the problem like this:

After I edited the file in LC, I converted with a converter (ODAFileConverter).

then netdxf reads your file without any problems.