haplokuon/netDxf #139: Coordinates Group

mlkmlt, 2019-10-25:
Hello, I am currently trying to group lines that are related
to define you as a group.

here's an example:

Example a rectangle:

1.50587453586086; 7.57888997459449; 0 'line 1 start
1.50587453586086; 2.74601133476647; 0 'line 1 end

1.50587453586086; 2.74601133476647; 0 'line 2 start
7.7841000586281; 2.74601133476647; 0 'line 2 end

7.7841000586281; 2.74601133476647; 0 'line 3 start
7.7841000586281; 7.57888997459449; 0 'line 3 end

7.7841000586281; 7.57888997459449; 0 'line 4 start
1.50587453586086; 7.57888997459449; 0 'Line 4 End

I would now like to group all lines based on their coordinates to calculate how many stitches I need.

my previous approach.

<Code>
Import netdxf
 Dim Coordinates List As New List (Of Vector3)
for each line data as Line in DXF_File.Lines
  Here you get all the properties that the line contains.
- Start point as Vector 3
- Endpoint as Vector 3
_Coordinate list.add (line data .StartPoint)
              _Coordinate list.add (line data .endpoint)
next line data
 Dim t1 = From g In coordinate list Select g.X, g.Y Group By X, Y Into Group
       For Each data In t1
                WriteLine (data)
        Next
</ Code>

Content of the coordinate list:

{X = 1,50587453586086, Y = 7,57888997459449, Group = VB $ AnonymousType_0`2 [System.Double, System.Double] []}

{X = 1,50587453586086, Y = 2,74601133476647, Group = VB $ AnonymousType_0`2 [System.Double, System.Double] []}

{X = 7.7841000586281, Y = 2.74601133476647, Group = VB $ AnonymousType_0`2 [System.Double, System.Double] []}

{X = 7.7841000586281, Y = 7.57888997459449, Group = VB $ AnonymousType_0`2 [System.Double, System.Double] []}

If I now filter the t1 on, it groups either only X or Y only.

X = 1.50587453586086, Y = 7.57888997459449
X = 1.50587453586086, Y = 2.74601133476647
X = 7.7841000586281, Y = 2.74601133476647
X = 7.7841000586281, Y = 7.57888997459449

If I now group X, it would be like this:
X = 1.50587453586086, Y = 2.74601133476647
X = 7.7841000586281, Y = 7.57888997459449

And so I have 2 groups that I can no longer group, the same goes for Y.

I do not know how I now s.o. as a group can summarize.

haplokuon, 2019-11-18:
Sorry but I do not understand what is the issue you have using this library.

mlkmlt, 2020-06-24:
Hey haplokuon,

i try to combine a Quader with 4 single Lines to 1 Group.

have you a Idea?

i try the Algorithmus fromTarjan.