haplokuon/netDxf #143: Delete layer with blocks and attributes

JoramQ, 2019-11-14:
I'm trying to delete layers and have a problem with block attributes. 
`List<DxfObject> entities = dxf.Layers.GetReferences(layer.Name);
foreach (DxfObject o in entities)
{
dxf.RemoveEntity(o as EntityObject);
}`

entities contains 3 items
- ATTDEF
- ATTRIB
- Insert

The ATTDEF and ATTRIB cannot be deleted because this is not an EntityObject so that will result in NULL. Then the layer cannot be removed because it still contains references. How to delete the attribute so the layer can be deleted?

haplokuon, 2019-11-18:
If everything goes as planned, to delete attributes (ATTRIB) you need to delete the Insert (Block reference) that owns them. To delete attribute definitions (ATTDEF) you need to remove the block since they belong to it, but to be able to delete a Block you must first remove all Insert entities that reference it.

JoramQ, 2019-11-20:
But in QCAD its possible to have a block without the attribute. If I select the ATTRIB there I can just delete it.

haplokuon, 2019-11-30:
Do not confuse Inserts and Blocks, an Insert is a block reference and it contains Attributes (ATTRIB) a Block is its definition and contains Attribute Definitions (ATTDEF). A Block does not contain Attributes.

You cannot delete Attributes from an Insert if this is what you are asking. It would be possible to add that functionality but, under normal circumstances, they are always linked to the Attribute Definitions of its Block. Keep in mind that you can always turn its Visibility off if you do not want to see them.