haplokuon/netDxf #145: Same name Problem when removing layout

TimmmiT, 2019-11-21:
If I want to remvo a layout which isn't the first in the list (Layout.AssociatedBlock.name != *papaer_Space) then the problem appears "There is already another block with the same name.".

The problem starts in the class Layouts, Method Remove by renaming the paperspace block names.
If i remove the secound layout the name *papaer_Space is still in the list, but the code trys to set the same name for another element.

`foreach (Layout l in this.list.Values)
            {
                // rename the corresponding PaperSpace block
                if (l.IsPaperSpace)
                {
                    string spaceName = index == 0 ? Block.PaperSpace.Name : string.Concat(Block.PaperSpace.Name, index - 1);
                    l.AssociatedBlock.SetName(spaceName, false);                  
                    index += 1;
                }
            }`

Maybe the problem can be solved by reversing the layout list. 
Or by renaming the Layout.AssociatedBlock.names first by using names outside the autocad naming convention.

Timm

TimmmiT, 2019-11-29:
ANother maybe the same problem:

I starts with an importet *.dxf.
Inside AutoCAD I manually created layouts, after i Made a mistake I deleted one layout in this *.dxf and created another new one. 
Now I see that in my *.dxf Model_Space, Paper_Space, Paper_Space0 and Paper_Space2 existst but Paper_Space1 is missing.  
Inside my code I want to clone one of these layouts. Inside your add method you creat a new Paper_Space and add the indexnumber by counting the List -2. So the new Paperspace gets the name Paper_Space2 (Which already exists!).
(See in your layouts class, layout.Add method, Line 132).

So in all these methods which creats, adds or remove Layouts (Paperspaces) you have to check if the Paperspace name is already given.

Yours sincerely
Timm

TimmmiT, 2019-11-29:
At the moment I have this solution:
`        //original Code
	//string spaceName = this.list.Count == 1 ? Block.DefaultPaperSpaceName :  string.Concat(Block.DefaultPaperSpaceName, this.list.Count - 2);

	//new Code by Timm Lemke (intermetric GmbH)
	string spaceName = GetPaperSpaceName();`

`private string GetPaperSpaceName() {
			string newPaperSpaceName= Block.DefaultPaperSpaceName;
			List<string> paperSpaces = new List<string>();

			foreach (Layout layout in this) {
				paperSpaces.Add(layout.AssociatedBlock.Name);
			}

			int i= 0;
			while (paperSpaces.Contains(newPaperSpaceName)) {
				i++;
				rueck = string.Concat(Block.DefaultPaperSpaceName, i);
			}

			return newPaperSpaceName;
		}`

TimmmiT, 2019-11-29:
At the moment I have this solution:
```
//original Code
//string spaceName = this.list.Count == 1 ? Block.DefaultPaperSpaceName : string.Concat(Block.DefaultPaperSpaceName, this.list.Count - 2);
```

```
//new Code
string spaceName = GetNewPaperSpaceName();

```

```
private string GetNewPaperSpaceName() {
    string newPaperSpaceName(= Block.DefaultPaperSpaceName;
    List<string> paperSpaces = new List<string>();

    foreach (Layout layout in this) {
        paperSpaces.Add(layout.AssociatedBlock.Name);
    }
    int i= 0;
    while (paperSpaces.Contains(newPaperSpaceName)) {
        i++;
        newPaperSpaceName= string.Concat(Block.DefaultPaperSpaceName, i);
    }
    return newPaperSpaceName;
}
```

Yours sincerely
Timm

haplokuon, 2019-12-03:
This should work in your end with the latest commit October,9,2019
```c#
DxfDocument doc = new DxfDocument();
doc.Layouts.Add(new Layout("Layout1"));
doc.Layouts.Add(new Layout("Layout2"));
doc.Layouts.Remove("Layout1"); // you can replace the string with any other layout name
doc.Save("test.dxf");
```	
The origin of the problem looks like is not within the layout removing process but with the layout associated block naming. AutoCad does not seems to care if a *Paper_Space0, *Paper_Space1,... is missing; but the *Paper_Space block must always be present (legacy code, I guess, long time ago there use to be only one paper space). If you delete a Layout associated with one of those 0, 1,... blocks it just delete it; but if the Layout associated with the *Paper_Space block is deleted, one must become the *Paper_Space, an existing one is renamed or a new one is created if that is the only one.

To accommodate to this, I am adding a new Layout if there are none when the file is saved, or just renaming the existing ones to strictly following the naming convention *Paper_Space, *Paper_Space0, *Paper_Space1,... But when adding a new layout to the list, the number of layouts is used to create the block name for the layout, and this might be the origin of your problem, if, for example, like in your case *Paper_Space1 block name is missing the new generated layout name will interfere with the existing one. You will end up with two layouts pointing to the same block and this is wrong, it might end up generating the issue you are experiencing. This is easily solved renaming the layouts during the DXF loading process, I will add a fix for that.

TimmmiT, 2019-12-04:
thank you for the quick reply.
Timm

closed 2019-12-04 by TimmmiT