haplokuon/netDxf #147: Cant Copy to Clipboard in AutoCAD

Pietervdwest, 2019-12-07:
If I generated \ save files in any version higher than 2004 then when the file is opened in AutoCAD and you want to copy the entities to the Clipboard (Ctrl - C) you get an error saying "Copy to clipboard failed". 
When saving the file as 2000 or 2004 format everything is fine and copy to clipboard works.

[image]

haplokuon, 2019-12-09:
I have no clue. The only important difference in the DXF file format of 2004 and earlier, is that it is saved as ASCII; later DXF file versions are saved as UTF-8; I do not see how this might affect the process of copying to the clipboard in AutoCAD once the file has been loaded.

From the flat look of the error message in the image seems like you are using Windows 10 (8, or 8.1 maybe), I remember another user having some kind of problem related with the styles and the fonts, when using an old AutoCAD version with Windows 10, a version not ready for it. Might your case be somewhat related to this?

I have test this with Windows 7 and I had no issues like you describe.

Pietervdwest, 2019-12-09:
Hi thanks for getting back to me. I have found a workaround. Let me explain and maybe that will shed some light on what it could be on your side

My process is as follows:

1.	I open a sheet metal file with Autodesk Inventor (a 3D mechanical CAD application)
2.	DXF file of the flat pattern profile is exported – to the correct DXF file version selected
3.	It is then opened in my application for basic processing – Add text and basic line entities
4.	The file is then saved

I have tested to ensure that it's not the Inventor application creating “bad” dxf’s but its not. I have exported all DXF versions from Inventor and they worked 100% in AutoCAD
To test, I have also stopped adding anything to the dxf file and I literally just opened the DXF file with netDXF and saved it again and straight after that, I had an issue in AutoCAD with those newly created DXF files.

So I then changed the process to do the initial export of the DXF file to 2000 dxf version. I then do the post-processing and then I use netDXF to create the DXF file in the correct version I require.

That has solved my problem!

I have added two files as exported from "Autodesk Inventor". They have no post-processing done on them yet. Like I said in my testing I only have to open and save them and then the 2007 one has issues.
The 2000 version file I can open and save in 2007 or any other format using netDXF, and all is fine.

Let me know what you find ok.

[DXF.zip](https://github.com/haplokuon/netDxf/files/3941127/DXF.zip)

haplokuon, 2019-12-11:
I have found the problem. There are two header variables $INTERFEREOBJVS and $INTERFEREVPVS that store handles to information related to the visual styles in AutoCad. The visual styles are not supported by netDxf and, although, the files where loaded without errors in AutoCAD, those two header variables, somehow, interfere with the copy to clipboard. In the next update I will avoid loading those header variables, since they contain data related with the visual styles and that information is not handled by netDxf, they are irrelevant.

This two header variables only appear in the 2007 and later DXF versions, I guess that is when Autodesk introduced the visual styles, I cant remember when was that. Anyways, at the moment and while you wait for the next update you can manually delete those variables form the CustomVariables of the document.DrawingVariables with `public bool RemoveCustomVariable(string name)`.

Pietervdwest, 2019-12-11:
Aaaah fantastic! Thank you so much and I am glad I could contribute something to this great tool! Keep up the good work mate!
Yes, for now, this is not a problem as I just keep saving the file out as 2000 DXF version and then convert to the required version using netDXF.

Thanks man you are a star!

closed 2020-09-03 by haplokuon