haplokuon/netDxf #152: [Question] Any method to obtain Bounding box of a block or DIM Block

alison84r, 2019-12-20:
Hello Team,
I have one question. Is there any way I can get directly bounding box of the BLOCKS, Dimensions or Its respective blocks. 
Or do I have to calculate? If yes what are the parameters to consider for calculation of bounding box.
[image]
[image]

Let me know if my question is not clear. 
Thanks in advance.

haplokuon, 2019-12-21:
No. There are some entities I cannot calculate it, like Text, MText, and Tolerance; I have no way of measuring strings based in TTF and SHX fonts. Also the Underlay and Shape entities have its size defined by the scale of its original dimensions that comes from the PDF, DWF, DGN, and SHX file.

LeiYangGH, 2023-05-07:
unfortunately!