haplokuon/netDxf #155: Possible Memory Consumption

mruch2, 2020-01-08:
I am planning on using this to open and import thousands of DXF's per week. This will run as a service. I will be loading the DXF's to extract the MText objects and then no longer need the DXFDocument. How do I dispose of the document in this scenario?

Any code examples?

I do not see an IDisposable implementation.

haplokuon, 2020-01-09:
There is no need for the DxfDocument class to implement the IDisposable interface. It is only required for types that need to release unmanaged resources, and this is not the case. The Net's garbage collector takes care of the allocation and release of memory for your application.

The only available examples can be found in the TestDxfDocument project.

mruch2, 2020-01-09:
Thanks for the response. I will check out the test examples.