haplokuon/netDxf #156: Can you please update the nuget package to version 2.3

Polymaker, 2020-01-13:
I don't know if you have stopped supporting the nuget package but the latest version is 2.2.0.1 and it misses a couple of useful fixes and features.
It would be very nice if you could upload the latest version to nuget.

Thanks.

haplokuon, 2020-01-15:
I do not publish anything outside what is available here at GitHub, you will need to contact the person who uploaded it at nuget.

Here, under the Code releases tab, you will find compiled binaries for this project.

Polymaker, 2020-01-15:
My bad. I saw your name under "Author" so I assumed that it was you that published on nuget but now I've checked again and I see that there is a different user under "Owner".

I'll stick to the Github releases then. Thanks for your answer.

closed 2020-01-15 by Polymaker