haplokuon/netDxf #157: [question] save dxf with zoom extents

sn0rkyITA, 2020-01-20:
hi
I'm looking for a method to save dxf with zoom extents....

at the moment I have this codes:

dxf.Viewport.ViewCenter = New Vector2(midX, midY)    'midpoint from a list of points
dxf.Viewport.ViewHeight = maxD             'maximum two points distance from a list of points

but it doesn't work as it should (viewcenter is not where it should be)

thanks

haplokuon, 2020-01-25:
Do not ask me too many details about this, I am just reading and writing those parameters from the DXF. There are a lot more stuff to do to implement a proper zoom extends. You have a combination of WCS (World Coordinate System) and DCS (Display Coordinate System) values. The DCS parameters will depend on the actual window in the UI. How all this play together? You will need to dig it up yourself.

sn0rkyITA, 2020-01-25:
:D ok thanks!