haplokuon/netDxf #158: Transformby for all entities (for scale)

sverduyn, 2020-01-22:
Hi,

I am trying to do the following (Autocad) using netdxf 
// Build a transformation matrix
            Matrix3d xform = Matrix3d.Scaling(25.4, Point3d.Origin);
            DocumentLock loc = doc.LockDocument();
            using (loc)
            {
                ed.WriteMessage("\nIn scale drawings, locked");
                // Start a Transaction
                using (Transaction tr = db.TransactionManager.StartTransaction())
                {
                    ed.WriteMessage("\nIn scale drawings, transaction");
                    // Open the drawing ModelSpace BlockTableRecord
                    BlockTable bt = (BlockTable)tr.GetObject(db.BlockTableId, OpenMode.ForRead);
                    BlockTableRecord mspace = (BlockTableRecord)tr.GetObject(bt[BlockTableRecord.ModelSpace], OpenMode.ForRead);

                    // Iterate through each entity
                    foreach (ObjectId id in mspace)
                    {
                        // Open the Entity for write
                        //ed.WriteMessage("\nIn scale drawings, Foreach");
                        Entity ent = (Entity)tr.GetObject(id, OpenMode.ForWrite);
                        // Scale the Entity
                        ent.TransformBy(xform);
                        ed.WriteMessage("\nIn scale drawings, after transform");
                    }

                    // Commit the changes
                    tr.Commit();
                }
            }
I have tried the following
Dim block As Block = Block.Load(item)
                Dim insert As Insert = New Insert(block) With {
                    .Scale = New Vector3(25.4)
                }
                Dim explode As List(Of EntityObject) = insert.Explode()
                Dim doc As DxfDocument = New DxfDocument()
                doc.AddEntity(insert)
                doc.AddEntity(explode)
                doc.Save(strFileToExport)
as well as 
Dim xform = Matrix3.Scale(25.4)
                Dim oDxfDoc As DxfDocument = DxfDocument.Load(item)
                Dim block As Block = oDxfDoc.Blocks.Item("*Model_Space")

                For Each blockRecord As EntityObject In block.Entities
                    blockRecord.TransformBy(xform, Vector3.Zero)
                Next
                oDxfDoc.Save(strFileToExport)

I have attached the before and after dxf's
[PartsBook - METRIC.zip](https://github.com/haplokuon/netDxf/files/4095704/PartsBook.-.METRIC.zip)

Thanks,

Shaun

haplokuon, 2020-01-25:
Are you using the latest code? Try it, I added a couple fixes that affect transformations of Ellipse Arcs since the v2.3 release.

sverduyn, 2020-01-25:
Hi,

Thanks for responding. Yes I am using the latest code.

haplokuon, 2020-01-26:
Try it now, hopefully this will be the last time I have to fix the ellipse arcs.

sverduyn, 2020-01-26:
I'm sorry to say that it is not the last time :-(. The entities are not being scaled at all now.

haplokuon, 2020-01-26:
The modifications I did do not affect the scale but the start and end angles of the ellipse arc. This piece of code works as intended and it does what you are trying to do. In red you will see the original drawing, in white the entities that has been scaled.
```c#
DxfDocument doc1 = DxfDocument.Load("0036-18-1003_PartsBookIso_BottomLeft_5.dxf");
DxfDocument doc2 = new DxfDocument();
Matrix3 scale = Matrix3.Scale(25.4);

// iterate through all entities in the "Model" layout, it is the same as
// foreach (EntityObject e in doc1.Blocks["*Model_Space"].Entities)
foreach (EntityObject e in doc1.Layouts["Model"].AssociatedBlock.Entities)
{
    // copy the original entity and scale it
    EntityObject clone = (EntityObject) e.Clone();
    clone.TransformBy(scale, Vector3.Zero);
    // add the entity to the new document
    doc2.AddEntity(clone);

    // copy the original entity and change its color to red
    EntityObject original = (EntityObject) e.Clone();
    original.Color = AciColor.Red;
    // add the entity to the new document
    doc2.AddEntity(original);
}
doc2.Save("test.dxf");
```
Double check your code, your drawing also contains lines, and it is not possible that suddenly both ellipses and lines are not scaled since I only modified the ellipse code.

sverduyn, 2020-02-02:
Apologies for my late reply, thank you for the code you gave. It seems I was still using the old dll instead of the new compiled dll. Sorry for the hassle :-(. Thanks for a great API and keep up the good work.

Shaun

closed 2020-02-02 by sverduyn