haplokuon/netDxf #160: "There is already another block with the same name."

johnnyontheweb, 2020-02-03:
I get a new error when using the commits by Dec 2019 until today:
"There is already another block with the same name."

This never happened before. Here is the stack:
`>	netDxf.dll!netDxf.Collections.BlockRecords.Item_NameChanged(netDxf.Tables.TableObject sender, netDxf.Tables.TableObjectChangedEventArgs<string> e) Riga 176	C#
 	netDxf.dll!netDxf.Tables.TableObject.OnNameChangedEvent(string oldName, string newName) Riga 50	C#
 	netDxf.dll!netDxf.Tables.TableObject.SetName(string newName, bool checkName) Riga 180	C#
 	netDxf.dll!netDxf.Blocks.Block.SetName(string newName, bool checkName) Riga 566	C#
 	netDxf.dll!netDxf.Collections.Layouts.RenameAssociatedBlocks() Riga 234	C#
 	netDxf.dll!netDxf.IO.DxfReader.PostProcesses() Riga 10007	C#
 	netDxf.dll!netDxf.IO.DxfReader.Read(System.IO.Stream stream, System.Collections.Generic.IEnumerable<string> supportFolders) Riga 284	C#
 	netDxf.dll!netDxf.DxfDocument.Load(string file, System.Collections.Generic.IEnumerable<string> supportFolders) Riga 851	C#
 	netDxf.dll!netDxf.DxfDocument.Load(string file) Riga 828	C#
`

This seems to be related to the issue 145, but still present.

haplokuon, 2020-02-06:
From the looks of your post seems that you have problems loading a DXF file, upload it I need to be able to reproduce the error.

johnnyontheweb, 2020-02-07:
Thanks for the reply. Please find attached a file
[dxfTest.zip](https://github.com/haplokuon/netDxf/files/4169848/dxfTest.zip)
 for testing.

closed 2020-09-03 by haplokuon