haplokuon/netDxf #161: DXF code 310

johnnyontheweb, 2020-02-05:
Hello,
I noticed that dxf code 310 is not supported. See:
http://help.autodesk.com/view/OARX/2018/ITA/?guid=GUID-3610039E-27D1-4E23-B6D3-7E60B22BB5BD

How to solve this?

haplokuon, 2020-02-06:
Where did you see that code 310 is supported?

johnnyontheweb, 2020-02-07:
I see it in many file: `if (this.code >= 310 && this.code <= 319) // string representing hex value of binary chunk`

Anyway, the error was related to wrong length of the subsequent binary field. I tried handling that by controlling the length of the field - if there's more than 255 chars, the previous code fails.
`        private byte[] ReadBytes(string valueString)
        {
            List<byte> bytes = new List<byte>();
            for (int i = 0; i < valueString.Length; i++)
            {
                string hex = string.Concat(valueString[i], valueString[++i]);
                byte result;
                if (byte.TryParse(hex, NumberStyles.AllowHexSpecifier | NumberStyles.AllowLeadingWhite | NumberStyles.AllowTrailingWhite, CultureInfo.InvariantCulture, out result))
                    bytes.Add(result);
                else
                    throw new Exception(string.Format("Value {0} not valid at line {1}", hex, this.currentPosition));
            }

            return bytes.ToArray();
        }`

If valueString.Length is >255, it fails. I know that in this case the file is not valid, but I kindly ask how I can overcome the issue - the only solution I found to skip a corrupted hex field is to return 0:
`            if (valueString.Length > 254) {
                byte[] bt = new byte[254];
                Array.Clear(bt, 0, 254);
                return bt;
            }`

haplokuon, 2020-02-12:
A couple of excerpts from the official DXF documentation.

From the "DXF Group Codes in Numerical Order Reference" codes 310 to 319 
"Arbitrary binary chunks with same representation and limits as 1004 group codes: hexadecimal strings of up to 254 characters represent data chunks of up to 127 bytes"

If the code 310 exceeds the limits the binary information must be divided into multiples 310 codes.

From the extended data information Binary data code 1004
"Binary data is organized into variable-length chunks. The maximum length of each chunk is 127 bytes. In ASCII DXF files, binary data is represented as a string of hexadecimal digits, two per binary byte"

For binary files:
"Extended data binary chunks (group code 1004) are represented as a single-byte, unsigned integer length, followed by the specified number of bytes of chunk data."

Also in the THUMBNAILIMAGE Section for the 310 code, it says: "Preview image data (multiple lines; 256 characters maximum per line)" Which contradicts the maximum number of characters established at 254 in the "DXF Group Codes in Numerical Order Reference". I always seen a maximum of 254 characters so this might be an error in the documentation. In a binary DXF the binary data is preceded by a byte that defines the number of bytes to follow, that will make 1+127 = 128 bytes maximum, what would equals to 256 characters in a text DXF, but the binary data in a text DXF is not preceded by the total bytes in the line and that will make a maximum of 254 characters. Perhaps, the error in the documentation comes from that.

Therefore and without getting more into the inner guts of the DXF, those codes, for text DXF files, must have an even number of characters and no more than 254, that equals to a maximum of 127 bytes per line. Why this limitations? I have no idea, It will save some space in the DXF, to write a byte you will need three characters but if you write them in hexadecimal you will only need two 16*16=256. You should only see characters from 0-9 and A-F, and never a "€". 

In any case, the codes 310-319 are never used anywhere in the library, they are not supported at the moment, so if you must need to avoid it you can just return 0 in:
```c#
this.code >= 310 && this.code <= 319) // string representing hex value of binary chunk
{
	return 0;
	// return this.ReadBytes(valueString);
}
```
Keep in mind that, while in a text DXF it is straight forward to just skip a line, it is not so much when dealing with binary DXFs. Mess up reading the exact number of bytes and the reader will fail.

johnnyontheweb, 2020-02-13:
Thank you so much, this is exactly what I needed.

closed 2020-02-13 by johnnyontheweb