haplokuon/netDxf #163: The LinetypeShepeSegment scale must be greater than zero

OakLD, 2020-02-20:
On some drawings, I get a strange error stating:

```
System.ArgumentOutOfRangeException: 'The LinetypeShepeSegment scale must be greater than zero.
Parameter name: scale
True value was 0.'
```

I was trying to debug it and modify the netDXF code to avoid the problem, but I was unsuccesfull. I don't really understand why it's happening and what source value is causing this problem, neither I could fix the consequences.

Any ideas?

haplokuon, 2020-02-22:
This comes from a complex line type, that seems it is not being used by any entity, Under normal circumstances values in a DXF that fall outside the valid range are given a default value; but sometimes someone sneaks past through. It is always much better to post the DXF file that gives you problems

OakLD, 2020-03-09:
I can confirm that the changes made in netDxf code are effective and it works well. Thank you very much.