haplokuon/netDxf #164: Import/export of dxf blocks with multiline attribute

Baryad, 2020-02-20:
Hello, haplokuon!
Could you help me with multi line attributes?
I'm trying to import the file with multi line attribute in a block, create few inserts and save the file.
It seems like multi line settings for attribute are not loaded nor saved.

In one dxf file I've created a block with a multi line attribute.
By netDxf I'm loading that block from the dxf, then creating a new dxf, inserting the block there with a multi line text.
In a new dxf text is not multi line.
How insert is generated sketch:
var dxf = new DxfDocument();
var srcDocument = DxfDocument.Load("MultilineAttributeInABlockSource.dxf");
var block = srcDocumentBlocks.FirstOrDefault(b => b.Name.Equals("Description"));
var insert = new Insert(block) {Position = new Vector3(0,0,0)};
insert.Sync();
insert.Attributes.FirstOrDefault(a => a.Tag.Equals("DESCRIPTION")).Value = "Multiline\\PText";
dxf.AddEntity(insert);
dxf.Save("MultilineAttributeInABlock.dxf");

The source dxf file https://github.com/Baryad/DxfSamples/blob/master/MultilineAttributeInABlockSource.dxf
The generated dxf file https://github.com/Baryad/DxfSamples/blob/master/MultilineAttributeInABlock.dxf

Thank you

haplokuon, 2020-02-22:
Multiline attributes are not supported. Someone might think that adding the newline symbol "\P" might be enough but the way they are implemented it is a lot more complex, it is actually a real mess that I have no plans to add it.

Baryad, 2020-02-22:
Thank you for the answer.
As a workaround  MText inside a block can be used. The specs for it is complex, especially the splitting it for separate attribute or text for older dxf versions.
Thank you for doing a great job implementing such complex format.

closed 2020-02-22 by Baryad