haplokuon/netDxf #165: Filter just Shape Lines by ignoring Dimension line from dxfDoc.Lines

aruyc, 2020-02-25:
[image]

Hello i have attached file for understanding my question.

Thank for such good library.
I want to differentiate that all dimension like lines and normal shape lines. How can i do that? I have checked all property of lines but unable to find difference between real shape lines and dimension or boundary lines. I get all lines as continuous in dxfDoc.Lines. I just want filter shape lines only. In attached file that tiny blue rectangle is then only one i required. Can you please help me.

haplokuon, 2020-02-27:
Lines like any other entity do not hold any kind of information about what are they connected with, so without a more deep analysis of the drawing it is impossible to know if a group of lines has the shape of a rectangle, a wall, a tree, or anything else. Therefore, when you need to analyze, programmatically, the contents of a drawing it is better to prepare before hand.

You have different options: you can assign a unique color, layer, line type to those entities so you can easily recognize them; you can also add extended data information to those entities, or you can select them through their handles if you know them. Probably, the way that will give you the most freedom is the use of extended data. It does not affect the appearance of the entity, and it can hold additional data if needed.

In your case you said that those lines are blue, you could check its color, but the color of an entity can be inherited from the layer they belong, in that case its color is defined as ByLayer.

aruyc, 2020-03-03:
OK. thank you so much for reply. I will try to differentiate extra lines(dimension table box lines and all) with Layer.