haplokuon/netDxf #166: DrawingUnits property make no changes in created dxf file

dataassemble, 2020-02-26:
Hi, I'm using a code example:
DxfDocument dxf = new DxfDocument();
dxf.DrawingVariables.InsUnits = DrawingUnits.Millimeters;
Line line = new Line(new Vector2(0, 0), new Vector2(10, 0));
dxf.AddEntity(line);
dxf.Save("test.dxf");
and my problem is that no matter if I change drawing units, line is always 10 inches long.
Is there a way to use millimetres and make that line of code (dxf.DrawingVariables.InsUnits) work for me? 
Thanks Leszek.

haplokuon, 2020-02-27:
The InsUnits value only is only applicable when you need to insert a drawing inside another one. It has the same function as the BlockRecord.Units property. The entities inside a drawing are not affected by it, if the length of a line is ten units, it is always ten units no matter what the value of the InsUnits is. If you need to scale the entities of a drawing when changing units you will need to do it manually.

dataassemble, 2020-02-27:
Ok thanks for clearing that out. So I want to save a line with length of 10 mm, and the way I can do it is by dividing 10 units of vector with 25.4 like in the code below:
DxfDocument dxf = new DxfDocument();
Line line = new Line(new Vector2(0, 0), new Vector2(10/25.4, 0));
dxf.AddEntity(line);
dxf.Save("Units.dxf");
My question, is there a way to set drawing units as millimetres, so I don't have to scale that 10 units of line with 25.4 factor?

haplokuon, 2020-03-01:
Sort answer is no.

Just be consistent and draw using the units you need, I do not see what is the problem. From the conversion factor you are using seems like your brain or your data works in inches but the drawing needs to be in millimeters. To help you with the conversion between units take a look at the Units namespace, specially the static class UnitHelper. You could even use the TransforBy method of the entity, but this method will be slower than just multiplying the coordinates of the line by a scale.

dataassemble, 2020-03-04:
Yes of course when we draw object we chose  units, but if person who don't know witch units was used in that drawing open it, can't be sure of a proper scale for that object except if I leave a legend inside. I have created a line in Solidworks with length of 10mm, exported as dxf file and I have also created a line 10 units in netDXF library like I mentioned before. After reviewing both files in notepad they have the same value of 10 under position "Model" and when I open them in eDrwings and measure, the Solidworks exported file have 10mm and netDXF 10 inch. I try to change a headers setting by adding $MEASUREMENT as 1 which means metric but it didn't change a thing. So that really confuse me as it thought it should be a way to define in file what units has been used. Thanks for your help.

closed 2020-03-04 by dataassemble

raysay, 2020-03-04:
抱歉，現在不方便接聽。