haplokuon/netDxf #170: Read multiple house types in a DXF file

yixingDBJ, 2020-03-09:
[image]
Hello, how about the DXF file like the picture? Can you identify several house types? If you can, which variable or method in dxfdocument

haplokuon, 2020-03-09:
Read [this](https://github.com/haplokuon/netDxf/issues/165), same problem.

yixingDBJ, 2020-03-10:
Thank you for your reply, but I may not have expressed my question clearly. For example, the result I want to get is that there is no way to automatically identify that it contains six independent parts, each part is a complete CAD picture, but they are placed on the same large picture

yixingDBJ, 2020-03-10:
[image]
At last, the effect of recognition is equivalent to that each of them is independent

haplokuon, 2020-03-14:
No. What to do with the contents of the DXF is up to you, but if you know the axis aligned bounding box of each part you can check which entities lies inside them.