haplokuon/netDxf #171: Text position and extrusion

Baryad, 2020-03-13:
Hello, Daniel!

I'm trying to add the text on Z axis up 400mm and on XZ plane
The version is netDXF.2.2.0.1 

var textEntity = new Text("test", Vector3(0, 400, 0), 20);
textEntity.Normal = Vector3(0,-1,0);
textEntity.Rotation = 0;

The position in dxf is not the same as expected
10
0.0
20
0.0
30
-400.0
and extrusion is correct
210
0.0
220
-1.0
230

In AutoCAD it's located at (0,400,0) instead on (0,0,400)
The expected values in dxf are
 10
0.0
 20
400.0
 30
0.0

(It looks like the same issue is with the circle transformations. Is it the same case?)

I'm using the version netDXF.2.2.0.1 

Thank you, Sergey!

haplokuon, 2020-03-14:
The text position as the circle center are defined in world coordinates, it is in the documentation. As you said you want the text at (0, 0, 400) WCS, therefore just pass that point to the Text constructor instead of (0, 400, 0). The DXF format stores those points in local coordinates, this is one of the many questionable decisions of the DXF specification.

Baryad, 2020-03-14:
Thank you, Daniel!
My bad! I had to read the documentation more carefully.
With the switching from OCS to WCS texts and circles are working as expected.

closed 2020-03-14 by Baryad