haplokuon/netDxf #176: Polyline entity is not recognized

ktheijs, 2020-03-31:
I have added a small dxf file that contains a single Polyline entity. It reads just fine when I use DWG TrueView, but no entities are found in the DxfDocument when I use netDxf. Does anyone know what the problem is here?

[NZ_transect.zip](https://github.com/haplokuon/netDxf/files/4407510/NZ_transect.zip)

lamber, 2020-04-01:
Hi ktheijs ,I faced the same problem,but solved it succesfully,you can get the Polyline datas in LwPolylines.

ktheijs, 2020-04-01:
Hi lamber, thanks for fixing it! Do you know in which version it will be released? Also, I saw that the latest stable version on NuGet is 2.2.0.1. Any idea when this gets updated?

lamber, 2020-04-10:
Hi ktheijs,I saw the lastest vesion is netDxf 2.3.0,what I need is just read the dxf files and get the model data to show the graph, I think this version has met my requirements.