haplokuon/netDxf #178: .CHM not working

zahmed80, 2020-04-06:
Hello, 

I downloaded the .CHM file, i can see the topics but there is no content 

kindly help

vsheveliov, 2020-04-06:
Try this way:
https://thirtysix.zendesk.com/hc/en-us/articles/202921675-How-to-Unblock-a-File-Downloaded-from-an-Email-or-the-Internet

haplokuon, 2020-04-07:
I don't know if the suggested tip has solved your issue, but I had a similar problem in the past with chm files that seemed to be related to the path folder, where the file is placed, contains non-ASCII characters, I cannot say if that includes also the extended characters.

The chm contains the same information as the xml that the Visual Studio intellisense uses to display the tooltips about the methods while you code. I used this [tool](https://github.com/EWSoftware/SHFB) to create the chm file.