haplokuon/netDxf #180: AlignedDimension labeling problem on version 2.3.0.0

0xblack, 2020-04-08:
When the Line used by the label is a line perpendicular to the X axis, that is, if the X coordinate values of the two endpoints of the Line are the same, the label will not be generated and affect other subsequent labeling operations

haplokuon, 2020-04-08:
Post a sample code so I can reproduce your problem. I need to know stuff like the dimension style used, any possible style overrides, if the block that represents the dimension is generated by netDxf (the DxfDocument.BuildDimensionBlocks property), etc.

Try to use the latest source code available, there are a couple fixes related to dimensions that perhaps are related with this issue.

0xblack, 2020-04-09:
Thanks for your attention and such a quick response, I have compiled a reproduction code as follows, reproduced on **version 2.3.0.0**
` public void AddAlignDimensionForQuestion()
            {
                  DimensionStyle dimensionStyle = new DimensionStyle("MytName")
                  {
                        DimScaleLinear = 1000,
                        LengthPrecision = 0,
                        TextHeight = 0.05d,
                        ArrowSize = 0.01d,
                        ExtLineExtend = 0.005d,
                        ExtLineOffset = 0d,
                        FitTextInside = true,
                        TextOffset = 0.005d,
                        TextVerticalPlacement = DimensionStyleTextVerticalPlacement.Outside,
                        DimBaselineSpacing = 0d
                  };
                  Vector2 pt1 = new Vector2(1, 5);
                  Vector2 pt2 = new Vector2(1, 6);

                  Line refLine = new Line(pt1, pt2);
                  AlignedDimension dimension = new AlignedDimension(refLine, 0.1);
                  dimension.Style = dimensionStyle;
                  Document.AddEntity(dimension);
            }`

haplokuon, 2020-04-11:
I haven't noticed any thing wrong the result of your sample, everything seems correct according to the dimension style values. What is the result you were expecting?

Just in case, try the latest code from the repository.

0xblack, 2020-05-17:
Sorry, this is a mistake of mine. The code works normally, and the result is correct. Thanks again for the help of the author

closed 2020-05-17 by 0xblack