haplokuon/netDxf #182: Dynamic Blocks

donnyv, 2020-04-16:
Did a test today using a dynamic block. I don't care about actions just want to get attributes out like a regular block. The name of the block was "DynamicBlockTest" but it reads in as "*U13". Also when I look at the insert.Layer.Name property it shows the layer name with a "  @ 40" at the end. I'm assuming this is just parsing issues with parsing the dxf file. 

Is there anyway around this or are you going to add at least partial support for dynamic blocks?

Thanks

haplokuon, 2020-04-18:
Read [this](https://github.com/haplokuon/netDxf/issues/103) about how dynamic blocks names. About that string " @ 40" added at the end of the Layer name, perhaps it is something related to a parameter in the dynamic block related to layers, I don't know, it is not a parsing issue. The main problem with dynamic blocks it is the complete lack of documentation about them, so do not expect to much support for them.

donnyv, 2020-04-18:
I understand now. Hopefully Autodesk will update the dxf spec with some info about dynamic blocks someday. Thanks!

closed 2020-04-18 by donnyv