haplokuon/netDxf #184: Hatches from intersection

mattiafedrigotti, 2020-04-20:
Sorry, but i have not found an answer in the issues.
Is possibile to draw an hatch from a intersection of bounadries? If I specify multiple boundaries for an hatches, how can I invert the selected area for the hatch?
Thankyou.

haplokuon, 2020-04-22:
When you have several boundaries that intersect each other, the hatch pattern will only be drawn where there is an odd number of superposition. This is the default behavior, to achieve any other result it is required to calculate the intersection, difference, or union; and that is not implemented.

I have somewhere, I will need to find it, some code that I programmed long time ago as an exercise to solve boolean operations, I did it from the description of a book that I do not remember at the moment. I could publish it but you will need to keep in mind that, it works only for polygons and triangle meshes, it is slow, and some glitches might appear for very small segments and triangles.

mattiafedrigotti, 2020-04-22:
Thanks for your answer.
I need only consider a even number of superposition > 0. Can I do this in your code?

haplokuon, 2020-04-23:
If you have two different intersecting boundaries and you want the hatch only in the intersection area, that intersection needs to be calculated, and this is something that the library doesn't do. The current implementation only consider cases of non overlapping boundaries, if the overlap the behavior will be as I explained in my previous post, there is no wary around that. If they do the result might not be what you expect it will depend on how the boundaries are defined, if they are part of the same HatchBoundaryPath, or different ones and its order.

The boundary data of a hatch in a case (overlapping boundaries) like this, will have only one boundary that reference two entities (in the case that the intersection shapes are two polylines) and an internal edge data that defines the intersection. For non associative hatches the referenced entities do not appear in the internal data of Hatch entity, just the edges.

mattiafedrigotti, 2020-04-30:
Can I insert a seed Point for selecting the boundaries intersection (code 98 in the dxf file)? This can be helpful.
Thank you very much.

haplokuon, 2020-05-01:
That will not work. The Hatch entity must contain the information of its perimeter that is defined by the edges of the boundary path.