haplokuon/netDxf #187: TableObjects limits

JoramQ, 2020-05-25:
The default limit of the TableObjects which is at int.MaxValue is a lot of times overwritten with short.MaxValue. Is there a reason for this?
I have for instance a DXF document with for some reason 34000 APPID ApplicationRegistry entries. It works if i remove the line "this.MaxCapacity = short.MaxValue;" but I'm not sure if that is a good idea.

An other solution could be to ignore the ApplicationRegistry entries that are not useful. But how to identify what is a standard Autocad and what is something that can be ignored?

haplokuon, 2020-05-26:
Yes, it is safe to remove or change its value if you need it. Those limits come from what the official DXF documentation says, these are some excerpts from it:

Common Symbol Table Group Codes
"70 Maximum number of entries in table"
The code group 70 corresponds to a 16-bit integer value. If it is signed or unsigned short I do not know.

In the About Symbol Table Group Codes (DXF) chapter it says:
"..., so do not use the count in the 70 group code as an index to read in the table. This group code is provided so that a program that reads DXF files can allocate an array large enough to hold all the table entries that follow."

I added those limits as a safety precaution according to the DXF documentation, I do not even know if they are still applicable or if surpassing that number can have side effects. For example, the number of control points of a Spline entity, if we take the DXF documentation by the letter, the maximum number is also a short (group code 73 - Number of control points). I actually tested this case and it can hold a lot more, if the number of control points surpass the limit of a short this group code will hold an invalid value. My guess is that the maximum number of table objects has a similar problem, but since I have not tested that the limits are still in place. If I had to bet, those limits come from a time memory was more precious, but when, exactly, those limits were lifted, I ignored. I might remove those limits from the TableObject collections and let it pass

Usually is a good practice to purge the drawings from leftovers once in a while, netDxf will read all table entries whether the are in use or not, it is up to you if you want to delete them.

JoramQ, 2020-06-08:
Thanks for the information.

closed 2020-06-08 by JoramQ