haplokuon/netDxf #188: Can we get the length and width of text and coordinate point information from DXF file?

lianzaozi, 2020-05-27:
Can netdxf parse the size and position coordinates of the text?

Can SHX and TTF fonts be parsed correctly?

haplokuon, 2020-05-28:
No. The only information available about Text and MText entities is what is stored in a DXF. There are no additional support to work with fonts.

lianzaozi, 2020-06-03:
In the future, does netdxf plan to add the height width and coordinate points of the vector font (.shx) characters, so that I can use the text stored in dxf for laser engraving and laser marking.

haplokuon, 2020-06-04:
No, I do not have nor make plans, first I will need to decipher how exactly the SHX works internally. I suppose to be a binary version of the SHP format, and I have not delved into it.