haplokuon/netDxf #190: Block Attributes Enumeration With Non Unique Tags

thewhitesnowman, 2020-05-29:
When attributes have non-unique tags the block attribute details (return from netdxf) cannot be trusted.  Is there another way to enumerate attributes?

Autodesk says:
Tag - The tag that identifies the attribute. Make sure that the tag name is unique.

Note: It is important to make sure that attribute tags have unique names. The Enhanced Attribute Editor will display any duplicate tags in red. Duplicate tags will cause problems when extracting data or if you use them in dynamic blocks.

haplokuon, 2020-05-30:
Despite what the AutoCAD documentation says it is possible to create a block with attribute definitions with the save tag. I restricted that option in netDxf, the main problem with that is when you create an Insert its attributes when saved to a DXF only holds the tag of its original attribute definition, therefore it is not possible to know which is which. It would have been a lot smarter from Autodesk to store its handle since they are unique, like they do in other places.

So, never use the same tag for different attribute definitions.

thewhitesnowman, 2020-05-31:
Thank you.  I'm working with old documents that are out of my control.  I spent a lot of time racking my brain about why my results were not as expected.

So, to confirm there is no work around to pull attribute data when tag value has been duplicated?

closed 2020-05-31 by thewhitesnowman

haplokuon, 2020-06-01:
Even if you will loose any attribute definition with duplicate tags from the block, you can still retrieve the full list of attributes from the Insert entity, regardless of they were created from a duplicate attribute definition from its referenced block. It is even possible that an insert attribute may point to a null attribute definition, in a DXF you can have attributes without an attribute definition.

thewhitesnowman, 2020-06-01:
I've got no problem retrieving values from blocks / inserts.

The prompt / value relationship is important (more so than tag, for me) to be sure I'm retrieving information that is usable.  One example is title block information...  I would like to rename the file based on title block details.

Thanks again for supporting and your great work with the library.