haplokuon/netDxf #193: Block Attribute

lemiroir, 2020-06-08:
When I change the visible property to false, I cannot find it in CAD software

lemiroir, 2020-06-11:
@haplokuon sorry,possibly I may not expressed my question clearly.I mean I set the block attribute visible property to false, I can't find it in CAD software. however, I create a block in CAD software with unvisible attribute, I can see the attribute in table after I insert to the document.

haplokuon, 2020-06-11:
What you need to do is to turn on the AttributeFlags.Hidden in the Attribute.Flags property. The IsVisible property, like any EnitityObject, just does that, it turns on or off the visibility of a graphical object. As far as I know, in AutoCAD the only way to change it is through code.

sn0rkyITA, 2020-08-29:
hi!
this is not correct, autocad show and empty attribute if you set IsVisibile property TRUE and if you try to edit block, the attribute just isn't there, isn't selectable...
however, a standard attribute inside a block have an "invisible mode" which you can use to add data without showing them on screen
you can see it with gestattbl command on autocad

a solution is needed
thanks

haplokuon, 2020-09-01:
That "invisible mode" of the attribute you are talking is the AttributeFlags.Hidden in the Attribute.Flags property I was talking in my previous post. The IsVisible property is a parameter in the DXF shared by all entities. My only guess for its existence is for plugins that need to keep stuff hidden from the user manipulation.

I do not see a problem there, the IsVisible property is always true by default. I have no idea about that "gestattbl" command you are talking about, it is not listed in the AutoCad commands list.

sn0rkyITA, 2020-09-01:
sorry, you are right! "hidden" is the right property