haplokuon/netDxf #200: get the drawing order

andreas-sommer-engenious, 2020-07-21:
I am no longer sure whether this library is still under development. I still ask a question. How do I get the drawing order out. The library categorizes the objects. I saw there is a hex handle attribut. Is this the drawing order?
Regards
Andreas

haplokuon, 2020-07-23:
The drawing entities in a DXF are ordered by Layout, therefore if you want to access the entities list as they were created use:

```c#
DxfDocument doc = DxfDocument.Load(@"sample.dxf");
foreach (Layout layout in doc.Layouts)
{
	foreach (EntityObject entity in layout.AssociatedBlock.Entities)
	{
		// the entities in the collection
	}
}
```
If you only need the entities in the model space the corresponding Layout is always called "Model".

You can also access the entities through the block list. The model space block is called "*Model_Space", paper spaces blocks are called "*Paper_Space", "*Paper_Space0", "*Paper_Space1", "*Paper_Space2", and so on.

andreas-sommer-engenious, 2020-07-24:
Hi,
Thanks you very much for your help. This works in general, but not for all dxf-files. I attached the special file. There you can see the hatches behind the three text symbol TOT. In Autocad, the hatch is behind the text and not behind all. Are there any commands saved in dxf?

[Schalter Hebel t-0-t 2xEin 12mm (100-000-026).zip](https://github.com/haplokuon/netDxf/files/4971119/Schalter.Hebel.t-0-t.2xEin.12mm.100-000-026.zip)

haplokuon, 2020-07-30:
They are in the order as they appear in the DXF section ENTITIES, and there your TEXT appear before the HATCH. I guess you have been using the Draw Order commands in AutoCad, It seems that those commands do not affect the order in which the entities are saved, but somehow it is kept when the file is saved and loaded in AutoCad, therefore that information must be saved somewhere somehow that I do not know at the moment.

One solution is cutting and pasting the entities you want in front, that will force a change in the entities list when saved to the DXF.