haplokuon/netDxf #203: Why my AlignedDimension not showing in AutoCad2010

KehindeIsijola, 2020-07-24:
I have this code to write a AlignedDimension that is not showing in AutoCad. This is the only entity not showing up. Please somebody should assist. 

createdxf.DrawDimensionbdV3(bds, new Line(p1, p2), m_dbTextSize, mydxf, AciColor.ByLayer, AYERSTYPE.layer_boundaryLine);

 public void DrawDimensionbdV3(string text, Line line, double th, DxfDocument dxf, AciColor iColor, LAYERSTYPE layer)
        {

            DimensionStyle style = new DimensionStyle("MyStyle")
            {
                DimArrow1 = DimensionArrowhead.None,
                DimArrow2 = DimensionArrowhead.None,
                TextOffset = 0.0,
                TextStyle = new TextStyle("Arial.ttf"),
                 DimLine1Off = true,
                DimLine2Off = true,
                TextHeight = th,
                
           };

            AlignedDimension dim = new AlignedDimension(line, 2.0, style )
            {
                Layer = dxf.Layers[mLayers[(int)layer].ToString()],
                Color = AciColor.ByLayer,
                UserText = text.ToString(),
                 Lineweight = Lineweight.ByLayer
                
                
            };
             dxf.AddEntity(dim);
        }

haplokuon, 2020-07-30:
I do not see anything wrong in the code you, that would make the dimension to not show. But, there are still several unknown variables that might affect the drawing.

KehindeIsijola, 2020-08-06:
Thank you for the comment.
I discovered the error in the format of the text.  I had
bds += "\\PX" + distance instead of just
bds += "\\X" + distance.
the P was inserted in error.
Your library is amazing.