haplokuon/netDxf #207: [Exception: The attribute definition cannot be added to the collection] when loading dxf file

eplesse, 2020-08-27:
Hello,
First congrats for your great lib which I have been using for a year now in order to instantiate autocad objects into 3d UNITY app.

I recently ran into this exception (using latest 2.4.0 release) when loading a dxf file (converted from a dwg file with AUTOCAD Architecture 2021).
Commenting lines 197,198 in "AttributDefinitionDictionnary.cs" helped me to bypass this exception and enabled me to load objects I needed. But of course this is no long-term solution.

I can provide you with the dxf file (which opens perfectly in AUTOCAD) but for confidentiality reasons, I cannot upload it here.

thanks. Emmanuel

haplokuon, 2020-08-28:
You can send me the DXF directly using the email you can find at the header of the source code files if you need to keep it private.

haplokuon, 2020-09-03:
I checked your file and...

This is an oversight on my part. Commenting those lines will avoid other conditions where an attribute definition cannot be added to the collection. Instead go to the Block class and in line 751, inside the "AttributeDefinitions_BeforeAddItem" method, delete the whole else if block:
```c#
else if (this.Name.StartsWith(DefaultPaperSpaceName)) // paper space blocks do not contain attribute definitions
{
	e.Cancel = true;
}
```

Paper space blocks can indeed contain attribute definitions, it is uncommon but there is nothing in the DXF format that forbids it.

eplesse, 2020-09-04:
Ok. This get rids of the problem and I can load all objects correctly. Thanks

haplokuon, 2020-12-06:
Resolved in 2.4.1

closed 2020-12-06 by haplokuon