haplokuon/netDxf #208: to use external block with attributes

sn0rkyITA, 2020-08-29:
Hi,
I have a question... is it possible to add an external block (already done for this purpose) to use it on the code? 
at the moment I have this code behind, can I change it with a call to external block-file and use attributedefinition directly on it?

```
For Each Punto In ListaPunti
                Dim block = New Block("PUNTO")
                Dim pointblock As Point = New Point(New Vector3(0, 0, 0)) With {
                .Layer = New Layer("PUNTI"),
                .Color = New AciColor(30)}
                block.Entities.Add(pointblock)

                Dim nome = New AttributeDefinition("NOME:")
                block.AttributeDefinitions.Add(nome)
                block.AttributeDefinitions("NOME:").Value = Punto.NOME.ToString
                nome.Position = New Vector3(offset, offset, 0)
                nome.Height = offset * 0.75
                nome.Layer = New Layer("Nome")
                ...

                Dim insert = New Insert(DirectCast(block.Clone(), Block), New Vector3(puntoo.Xcoord, puntoo.Ycoord, puntoo.Zcoord))
                insert.Sync()
                dxf.AddEntity(insert)
Next
```

I hope you understand my question
sorry for my basic english

haplokuon, 2020-09-01:
Yes you can. Check the Create and Load methods of the Block class. To add AttriubteDefinitions to the model space access the associated block of the model Layout. One way or the other in the DXF structure everything belongs to a block, the model Layout and paper space layouts all are associated with an internal block.  The model space block is called "*Model_Space", paper spaces blocks are called "*Paper_Space", "*Paper_Space0", "*Paper_Space1", "*Paper_Space2", and so on.

You can create your block through code save it and then load it from the DXF into a different document, if that is what you are trying to achieve. And don't forget to post your code if you find any errors during this process, that's is something I will never be 100% sure.

sn0rkyITA, 2020-09-01:
thanks a lot! it works fine!