haplokuon/netDxf #209: tableEntryXData.Add(vport, xData) throws System.ArgumentException: 'An item with the same key has already been added.'

precisionterrain, 2020-09-03:
In the DXF file I am reading there is an error being thrown in DxfReader.cs indicating a duplicate key.

[image]

the problem can be solved by expanding the offending line out to:
[image]

However, I am unfamiliar with the DXF format and do not know if this will only kick the can further down the road...

haplokuon, 2020-09-04:
The Model viewports are not implemented, and this issue comes from them. In any case this error should not appear, only the active viewport needs to be read. When multiple model viewports are present all of them are called *Active, and if two or more of them have extended data  that "tableEntryXData" will complain.

Doing what you suggest will combine all extended data information  into one viewport the only one that is implemented in the DxfDocument, a part from that, it should not create any more issues.

If you can post that DXF file, so I can take a look.

precisionterrain, 2020-09-04:
Ok. Thanks for looking at it. I've sent the file to an email I found for you in the source code...

haplokuon, 2020-12-06:
Resolved in 2.4.1

closed 2020-12-06 by haplokuon