haplokuon/netDxf #215: multi line from array list?

lesonglam, 2020-09-16:
Dear all,

I am a very new user of netdxf. I am wondering how to draw a sequence of lines from a array?
In stead of :
   ```
        netDxf.Entities.Line line1 = new Line(new Vector2(0, 0), new Vector2(10, 10));
            netDxf.Entities.Line line2 = new Line(new Vector2(10, 10), new Vector2(5, 20));
            DxfDocument doc = new DxfDocument();

            doc.AddEntity(line1);
            doc.AddEntity(line2);
```
or
    ```
       double[] x = new double[3];
            double[] y = new double[3];
           x[0] = 0; y[0] = 0;
            x[1] = 10; y[1] = 10;
            x[2] = 5; y[2] = 20;
            netDxf.Entities.Line line1 = new Line(new Vector2(x[0], y[0]), new Vector2(x[1], y[1]));
            netDxf.Entities.Line line2 = new Line(new Vector2(x[1], y[1]), new Vector2(x[2], y[2]));
            doc.AddEntity(line1);
            doc.AddEntity(line2);
```

I need to do something like creating 2-D array, type double; each new line will be added one by one in a loop.
For example: 
       ```
     double[] x = new double[3];
            double[] y = new double[3];
           x[0] = 0; y[0] = 0;
            x[1] = 10; y[1] = 10;
            x[2] = 5; y[2] = 20;
            netDxf.Entities.Line line[] = new Line[3];
           
          for (int i=0;I<3;i++)
          {
             line[i] = x[i] ,y[i];
          }
          doc.AddEntity(line);

```
Thank you in advance,
Lam

haplokuon, 2020-09-22:
I do not know if I really understand your question. If what you are trying to build a rectangular array, you will need two loops, one for counting the number of rows and another one for the columns. For each tick of the loop you will need to increment the distance between the columns and rows respectively. Once the start and end points has been calculated you can create the line and add it to the document.

You should be able to find a lot more details on how to do this kind of thing around the web.

lesonglam, 2020-09-23:
List<netDxf.Entities.Line> lnx = new List<netDxf.Entities.Line>();

                for (int k = 0; k < Hull.hull_concave_edges.Count; k++)
                {
                    double x1 = Hull.hull_concave_edges[k].nodes[0].x;
                    double y1 = Hull.hull_concave_edges[k].nodes[0].y;
                    double x2 = Hull.hull_concave_edges[k].nodes[1].x;
                    double y2 = Hull.hull_concave_edges[k].nodes[1].y;
                     lnx.Add(new Line(new Vector2(x1, y1), new Vector2(x2, y2)));
                }
I can use this code in order to put all the concave hull in a loop.
However, I am still struggling to find the set of polygon vertexes instead if set of lines.  For example, there are 4 lines, such as l1, l2, l3, and l4, getting from Hull.hull_concave_edges. From 4 lines, we can get 8 nodes, which include 4 of them are the same position. How do neglect these points and get 4 point  in a order ?
I am thinking about using a manual loops, but it seems like time consumpming.

haplokuon, 2020-09-24:
Let's see if I understand this correctly, you have some kind of method that calculates the contour of, for example, a point cloud. That method returns a set of edges defined by its start and end point, but you only want the vertexes of the contour polygon.

The first thing you need is to make sure that the edges returned from hull_concave_edges define a continuous contour, that is the end point of the edge k is the same as the start point of edge k+1. If this is your case you only need to read the start point of every edge, or the end, same thing.

Now, if the set of edges is not continuous you will have to reorder it. Pick one edge and check which is the one to follow from the rest of the set, you may even need to check both the start and end point of the next edge, in case the direction of the edges is not the same. Once it is found store the start point of the edge as in the previous paragraph. Proceed again with this newly found edge to find its follower, and so on until the loop is complete.