haplokuon/netDxf #216: LWPolyline.Reverse not working correct (bug)

spitfirekmt, 2020-09-16:
/// <summary>
        /// Switch the polyline direction.
        /// </summary>
        public void Reverse()
        {
            this.vertexes.Reverse();
        }

needs to be changed into something like:

LwPolyline _lwp = (DXFFile.LwPolylines[0]).Clone() as LwPolyline;
            lwp.Reverse();
            List<LwPolylineVertex> _oldvertexes = new List<LwPolylineVertex>(_lwp.Vertexes);

            _lwp.Vertexes.Clear();
            for (int i = 0; i < _oldvertexes.Count()-1; i++)
            {
                _lwp.Vertexes.Add(new LwPolylineVertex(_oldvertexes[i].Position, -_oldvertexes[i + 1].Bulge));
            }
            _lwp.Vertexes.Add(new LwPolylineVertex(_oldvertexes.Last().Position, -_oldvertexes[0].Bulge));

The reversed vertexes have to get the negative bulge of the next position. Atm the result has a high artistic value.

haplokuon, 2020-09-22:
Yes, the bulge values need to be moved one place backwards after reversing the list of vertexes.

haplokuon, 2020-12-06:
Resolved in 2.4.1

closed 2020-12-06 by haplokuon