haplokuon/netDxf #218: How to get the PolygonalVertexes of SPline witch is made by fit?

kchbpy, 2020-09-28:
Is there any way to get the PolygonalVertexes of spline(fit method)?

Or how to convert the Spline from fit method to control method (AutoCAD have this function)

haplokuon, 2020-09-29:
This is not implemented. You can find more information about how to calculate the control points from a set of fit points [here](https://www.geometrictools.com/Samples/Mathematics.html), check the sample "BSplineCurveFitter". There is also more information in the book "The NURBS book" by Les Piegl and Wayne Tiller, chapter "9.4.1 Least square curve approximation" page 410.

kchbpy, 2020-09-30:
> This is not implemented. You can find more information about how to calculate the control points from a set of fit points [here](https://www.geometrictools.com/Samples/Mathematics.html), check the [...]

Thanks for your reply