haplokuon/netDxf #219: DXF 2004

FilatovAV, 2020-10-07:
Unfortunately it doesn't read dxf files in 2004 format

FilatovAV, 2020-10-07:
in void BuildLayerStateManager
```
                                Layer layer = (Layer) this.doc.GetObjectByHandle((string) recordEntry.Value);
                                if (layer is null) 
                                { 
                                    enumerator.MoveNext();
                                    continue;
                                }
```

this will fix the situation

haplokuon, 2020-10-09:
Please post your DXF. I didn't see any problems with files saved as 2004 generated by a later version of AutoCad. From what you wrote seems that the handle of the layer is not found and that should not happen, errors with handles in a DXF usually ends up in a total crash when loaded in AutoCad. What program generated your file?

FilatovAV, 2020-10-09:
The file was saved in Autodesk AutoCAD 2017. But it doesn't really depend on the version.
[TEST.zip](https://github.com/haplokuon/netDxf/files/5354259/TEST.zip)

haplokuon, 2020-10-10:
I think you have uploaded the wrong file. It didn't give me any problems and it doesn't contain any layer state.

FilatovAV, 2020-10-10:
it's funny, but I can't read this file, or any file created in DXF 2004 format.

[TEST2.zip](https://github.com/haplokuon/netDxf/files/5358939/TEST2.zip)

FilatovAV, 2020-10-11:
and I check on different computers

FilatovAV, 2020-10-11:
the library cannot read the DXF 2004 format everywhere unless you make the above edits

haplokuon, 2020-10-12:
The problem here is not if that check is needed or not. I will add it just to avoid possible problems with bad handles, issue that it is the result of a bad DXF.

The problem here is that you are posting DXFs that do not contain any layer state, the dictionary is there but there are no XRecords that hold the layer state data. If there are no XRecords the code you are editing will not be executed , therefore your modifications should not fix anything. Something else must be happening here and that is what confuses me.

Try the attached DLLs with the DXFs files you have uploaded, they are compiled from the latest source code found in the repository. Let's see what happens, oh! and there is no need to change computers.

[netDxf latest.zip](https://github.com/haplokuon/netDxf/files/5363257/netDxf.latest.zip)

FilatovAV, 2020-10-12:
No changes, DXF 2004 does not read, save in DXF 2007 and everything is fine.

FilatovAV, 2020-10-12:
although now I downloaded the master branch, compiled the project and it read DXF 2004 normally.