haplokuon/netDxf #220: Upgraded 2.3 to 2.4 - now producing bad DXF

ausjef, 2020-10-07:
I was previously writing DXF files with version 2.3 but had issues that I could not identify - issues that prevented some DXF / CAD readers from displaying the file.  After reading the changelog for 2.4, I guessed the transparency was the problem.  Tried with version 2.4 - now none of my DXF / CAD tools can open the file. Those tools are QCAD, and CADDirect 2021.   I noticed the table headers 'AcDbSymbolTable' in the 2.4 version don't have any 70  then another integer following - the way the earlier version did.

haplokuon, 2020-10-20:
First things first. This is not an issue with the 2.4 version but with the addition of the layer states in the latest commit, If you need to know which release correspond to which source code check the Tags or the Releases section.

I do not know how many times I being looking into this just to find out it was as simple as, in the BeginTable of the DxfWriter instead of writing "this.chunk.Write(102, "}");" I wrote "this.chunk.Write(102, "{"); as simple as messing up the curly brackets. I will upload a fix for this in a quick update. Funny thing is that AutoCad didn't show any problems.

ausjef, 2020-10-20:
Thank you for looking into this - and thank you for providing this excellent programming tool with examples that cover just about everything.

haplokuon, 2020-12-06:
Resolved in 2.4.1

closed 2020-12-06 by haplokuon